plapointe6/EspMQTTClient #1: 128 byte payload limit

ntableman, 2019-01-10:
Anything over 128 bytes in the MQTT payload causes the system to fail.

Is there a way to raise this limit?

ntableman, 2019-01-10:
Ah perfect...I was tracing and tracing to find where it was set! Thanks!

closed 2019-01-10 by plapointe6

abqmichael, 2019-04-01:
Thank you for the clarification that there may be additional issues at play regarding the maximum payload. I noticed that MQTT_MAX_PACKET_SIZE is set to 128 in the PubSubClient.h I have, but even though the code has a comment "must correspond to MQTT_MAX_PACKET_SIZE of PubSubClient.h," that it's set to 256 in the current version of EspMQTTClient.h. 

Might it be more straight-forward to either use MQTT_MAX_PACKET_SIZE in your code or, alternatively, set them to be equal in EspMQTTClient.h? Or is there another thing at play?
 
// MQTT_MAX_PACKET_SIZE : Maximum packet size
//    must correspond to MQTT_MAX_PACKET_SIZE of PubSubClient.h
#define MAX_MQTT_PAYLOAD_SIZE MQTT_MAX_PACKET_SIZE

As an aside, doing this compiles on my computer BUT when I increase  MQTT_MAX_PACKET_SIZE in PubSubClient.h to 512, I can only send about 488 characters. I'm guessing that's because the rest of the MQTT packet takes up space.

Thank you.

plapointe6, 2019-04-01:
Hello,

First, thanks for the comment, I made some tests and it works fine if we set directly MAX_MQTT_PAYLOAD_SIZE  to MQTT_MAX_PACKET_SIZE of PubSubClient.h. I made the change (see commit 915d5b161bfd5387d236304baa732bd068aa37a4).

Second, I will do more testeing regarding the 488 max payload size and I will update you once I will be done. I have seen similar issues about that from library [PubSubClient](https://github.com/knolleary/pubsubclient), see [here](https://github.com/knolleary/pubsubclient/issues?utf8=%E2%9C%93&q=is%3Aissue+size)

plapointe6, 2019-04-01:
Here is the results of my tests.

First, I tested with MQTT_MAX_PACKET_SIZE of PubSubClient.h at 512 bytes. The short message and the 450 bytes message has been sent, but not the 750 bytes message.
After, I changed  MQTT_MAX_PACKET_SIZE to 1024 and it worked well for the three messages.

Results :
[image]

Sketch
```c++
#include "EspMQTTClient.h"

void onConnectionEstablished();

EspMQTTClient client(
  "x",          // Wifi ssid
  "x",   // Wifi password
  "x",  // MQTT broker ip
  1883,             // MQTT broker port
  "x",    // MQTT username
  "x",      // MQTT password
  "test2", // Client name
  onConnectionEstablished // Connection established callback
);

void setup()
{
  Serial.begin(115200);
}

void onConnectionEstablished()
{
  client.publish("mytopic/test", "This is a short message");
  
  client.publish("mytopic/test", "This is a 450 bytes message ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------end");

  client.publish("mytopic/test", "This is a 750 bytes message ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------end");
}

void loop()
{
  client.loop();
}
```

plapointe6, 2019-04-01:
I digged further into PubSubClient and I found that the MQTT_MAX_PACKET_SIZE is the maximum size of the whole packet. This include the headers (5 to 9 bytes) AND the size of the topic string where you want to send the message.

abqmichael, 2019-04-02:
Thank you

I put a thank you note in the issue discussion.

Michael

abqmichael, 2019-04-02:
Thanks for getting on this. 

I am switching over from using PubSubClient and like your setup a lot more. It also has he 128 byte limit that’s harder to resolve. So I needed/wanted to understand or resolve that right away. A thing I like about yours is related. I’m planning to make my ESP software generic (just do the job) and putting application logic in a more powerful server (like a Raspberry Pi or desktop machine). Seeing that message size limit was going to be an issue, I was moving toward using multiple topics to allow me to minimize message size. Your setup makes that much easier than PubSubClient as it lets you assign callbacks on a per-topic basis. That seems much more reasonable.

In this case, simplifying the path forward in the event I need bigger messages even with the topic-based messaging scheme gives me confidence that won’t be the issue it is with your library.

Michael