plapointe6/EspMQTTClient #2: Solving the problem with connecting to Wi-Fi

PiterKh, 2019-01-10:
Hello!
I had problems connecting to the network via Wi-Fi.
The solution was to add the WiFi.waitForConnectResult () function to the void EspMQTTClient :: connectToWifi () method.
Now it works for me like this:

void EspMQTTClient :: connectToWifi ()
{
WiFi.mode (WIFI_STA);
WiFi.begin (mWifiSsid.c_str (), mWifiPassword.c_str ());

if (mEnableSerialLogs)
Serial.printf ("\ nConnection to% s", mWifiSsid.c_str ());

WiFi.waitForConnectResult ();

mLastWifiConnectionMillis = millis ();
}

Best regards, Peter

PiterKh, 2019-01-10:
I use ESP-12F.

plapointe6, 2019-01-10:
WaitForConnectionResult() loop until the wifi status become something else than DISCONECTED (https://github.com/esp8266/Arduino/blob/fcf2ac5d3de6c17fe177e16ccd1a2789826c2aa0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L367)

I will need more details of the issues you encoutered as the goal of EspMQTTClient is to provide solution to handle WIFI and MQTT connection without blocking the entire sketch when connecting or when there is a connection lost. This allow other task to be done in your sketch wile connecting. For exemple, you could blink a led while waiting for a connection and more.

PiterKh, 2019-01-11:
Before:
[image]
[image]

After:
[image]
[image]

closed 2019-01-11 by PiterKh

PiterKh, 2019-01-11:
> WaitForConnectionResult() loop until the wifi status become something else than DISCONECTED [...]

Unfortunately, to solve the problem of blocking in this way will not work. In any case, the connection takes time, which blocks the main task loop. For critical tasks it is better to use the “time ticker” (https://circuits4you.com/2018/01/02/esp8266-timer-ticker-example/)

plapointe6, 2019-02-21:
With version 1.2, I think this issue will be fixed. Feel free to re-open the issue if it's not the case.