plapointe6/EspMQTTClient #3: Support to will Message

aaric-con, 2019-02-03:
Hi,

Do we have any support to will messages?

plapointe6, 2019-02-08:
Hi,

There is no support for will messages.
However, it is possible to send a message as soon as you are connected to the MQTT broker by doing it in the connection established callback (onConnectionEstablished in the exemple).

It's your responsability to send the teastament message as the last action in your sketch, but we can't handle a unplanned power loss.

Regards

closed 2019-02-08 by plapointe6

plapointe6, 2019-04-14:
Hi,
I was wrong in my last reply, it is possible to handle last will message. My bad.
I just added the support for last will message, see ce28a9e211bcf7fffcd28a19765978f45d995587
It will be bundeled in the next version.