plapointe6/EspMQTTClient #5: Unsubscribe function vs multiple handlers for the topic

grafspirt, 2019-03-03:
The subscribing function allows to have a number of handlers per one topic.
The receiving function works with the same assumption - there can be a number of callback functions (structures) for the same topic, so they all need to be executed.
But the unsubscribing function removes only the first found "subscription" record.
I guess this has to be like:
`mTopicSubscriptionList[i] = mTopicSubscriptionList[i + found_counter];` instead of `mTopicSubscriptionList[i] = mTopicSubscriptionList[i + 1];`
and
`mTopicSubscriptionListSize = mTopicSubscriptionListSize - found_counter;` instead of `mTopicSubscriptionListSize--;`.

Or remove this "multiplicity" with topics.

plapointe6, 2019-03-04:
Hello @grafspirt, thanks for raising this point, I totally agree with you.

I think that the best fix we can do is to remove the "multiplicity" with topics.
I dont think that it is usefull for anyone, it cause more issue than benefits.

We needs to change the behavor of the subscribe function to check if the topic already exist in mTopicSubscriptionList before subscribing to it.

It is a simple modification, but it needs to be tested, so I will probably change that in the next days. However, If you want to change it, feel free to do it :)

Regards.

plapointe6, 2019-03-06:
Resolved with pull request #6, thanks to @grafspirt :)

closed 2019-03-06 by plapointe6