plapointe6/EspMQTTClient #7: New release, cannot get example or old code to work

ntableman, 2019-03-13:
Can you guide me on what changed in this, so I cam update my code? Your example won't compile for me even.

grafspirt, 2019-03-13:
There was a change of the class constructor attributes sequence to make available short forms when only a minimum of necessary attributes may be passed. Like:
```
EspMQTTClient client(
  "ssid",                 // Wifi ssid
  "pass",                 // Wifi password
  onConnectionEstablished,// MQTT connection established callback
  "ip"                    // MQTT broker ip
);
```

The long form taking all attributes has become as follows:
```
EspMQTTClient client(
  "ssid",                 // Wifi ssid
  "pass",                 // Wifi password
  onConnectionEstablished,// Connection established callback
  "ip",                   // MQTT broker ip
  1883,                   // MQTT broker port
  "mqttusr",              // MQTT username
  "mqttpass",             // MQTT password
  "test",                 // Client name
  true,                   // Enable web updater
  true                    // Enable debug messages
);
```
The attached to the package example code initiates the class object this way and is compiled ok.
Please have a try with the an example from the branch and report a problem if any appears.

grafspirt, 2019-03-18:
If there are no objections and new inputs I assume to close the topic.

plapointe6, 2019-03-20:
Sorry for late response, I had health problems :(
I made minor changes to prepare for v1.3 release.
One of them is to support old constructor formulation (see commit 4ea0455957734a6554bcf8961c618f38f57406e6)

Regards

closed 2019-03-20 by plapointe6