plapointe6/EspMQTTClient #8: HELP:I need more MQTT to receive the information

zsq19958629, 2019-03-16:
// Subscribe to "mytopic/test" and display received message to Serial
  client.subscribe("mytopic/test/+", [](const String &payload) {
   //Serial.println(topic);-------------------->This is the information I need,and more
    Serial.println(payload);
  });

How do I get more information parameters returned by the MQTT server?
thanks!!!

grafspirt, 2019-03-18:
Current approach of the library suggests to have separate callback function per each topic. So you can understand what topic has been posted by the particular callback is being triggered.
As soon as this library is based on [pubsubclient](https://pubsubclient.knolleary.net/) library, so data which may be received from MQTT message is limited by topic and payload. What kind of additional information do you need (it would be better to have references on http://mqtt.org/)?

zsq19958629, 2019-03-18:
My English is very poor, the following content is the machine to translate Chinese to English

My device is based on Espunra firmware and has 4 channels of relays.

Control with “ESPURNA-D34ECA/relay/set/0，0”to OFF，“ESPURNA-D34ECA/relay/set/0，1”to ON。

The MQTT server returns the “ESPURNA-D34ECA/relay/0，0 or 1” flag status。
ex：
MQTT - Message received [ESPURNA-D34ECA/relay/0] 0 
MQTT - Message received [ESPURNA-D34ECA/relay/1] 0 
MQTT - Message received [ESPURNA-D34ECA/relay/2] 0 
MQTT - Message received [ESPURNA-D34ECA/relay/3] 0
I need to determine which pin I control by the topic and payload that the MQTT server returns.

I need to subscribe to the topic "espurna-d34eca /relay/+", subscribe to all relay states of the device, and then control the corresponding port pin of Esp8266 according to the corresponding state. The +" sign is a wildcard that represents a subscription to all messages under the topic of "relay," excluding child messages.

However, the library does not support the wildcard "#" or "+" characters because the following code in the code limits the topics received.
  // Send the message to subscribers
  for (int i = 0 ; i < mTopicSubscriptionListSize ; i++)
  {
    Serial.printf("MQTT - Debug1:[%s] %s \n", topic, payloadStr.c_str());
    if (mTopicSubscriptionList[i].topic.equals(topic))
      {
        Serial.printf("MQTT - Debug2:[%s] %s \n", topic, payloadStr.c_str());
        (*mTopicSubscriptionList[i].callback)(payloadStr); // Call the callback
      }
  }
Debug2 can never output any information while using wildcards。

This library is the easiest I've used for MQTT. I hope it will be better in the future!

thanks！

zsq19958629, 2019-03-18:
Or can I subscribe to different topics multiple times?
ex:
  client.subscribe("ESPURNA-D34ECA/relay/0", [](const String &payload) {
    Serial.println(payload);
   //Execute my control code 0
  });
  client.subscribe("ESPURNA-D34ECA/relay/1", [](const String &payload) {
    Serial.println(payload);
   //Execute my control code 1
  });
  client.subscribe("ESPURNA-D34ECA/relay/2", [](const String &payload) {
    Serial.println(payload);
   //Execute my control code 2
  });
  client.subscribe("ESPURNA-D34ECA/relay/3", [](const String &payload) {
    Serial.println(payload);
   //Execute my control code 3
  });

zsq19958629, 2019-03-18:
> Or can I subscribe to different topics multiple times?
> ex:
> client.subscribe("ESPURNA-D34ECA/relay/0", [](const String &payload) {
> Serial.println(payload);
> //Execute my control code 0
> }); [...]

It works, I'm sorry, my mind stops working.

closed 2019-03-18 by zsq19958629

grafspirt, 2019-03-18:
Or the following variation of the approach you have described above (to have more easy-read code):

```c++
void HandlerPin1(const String & payload) {
  // Code for the topic 1
}

//...

void on ConnectionEstablished() {
  client.subscribe("ESPURNA-D34ECA/relay/1", HandlerPin1);
  //...
}
```

zsq19958629, 2019-03-18:
> Or the following variation of the approach mentioned above (to have more easy-read code):
> 
> ```c++
> void handlerPin1(const String & payload) {
>   // Code for the Pin 1
> }
> 
> //...
> 
> void onConnectionEstablished() {
>   client.subscribe("ESPURNA-D34ECA/relay/1", handlerPin1);
>   //...
> }
> ```

Thank you very much. I'll try your code style later.