plapointe6/EspMQTTClient #9: onConnectionEstablished setting

vbzc, 2019-04-14:
The onConnectionEstablished setting is removed in the new version. If I need to define two Mqtt objects in the sketch, how do I write onConnectionEstablished code for these two objects? Thank you

plapointe6, 2019-04-14:
Hi @vbzc 

Can you send me a sample sketch please ?
I want to figure out how it is possible to use two EspMQTTClient in the same sketch ?

It will help me to figure out the best fix for this.

Regards.

vbzc, 2019-04-16:
Hello, I sent you an email.

plapointe6, 2019-04-16:
Ok, I see what you want to acheive :)

You want to communicate with two different MQTT broker while being connected to the same WIFI network.

Doing so is a bit unclean with EspMQTTClient as is. It duplicates the wifi handling.

Let me think about the best way to handle that propelly inside the library, I will update when it will be ready for testing.

vbzc, 2019-04-16:
thank you very much.

Regards

plapointe6, 2019-04-17:
I made changes in the `dev` branch to allow the use case referenced in this issue plus some memory optimisations. If you are willing I suggest you to try the new changes and let me know if it worked.

See https://github.com/plapointe6/EspMQTTClient/tree/dev for more details.
I've written an exemple to illustrate how to use two MQTT connections in the same sketch.

Take note that I plan to make changes in the `dev` branch soon (for ESP32 support), it is possible that this branch will be less stable.

vbzc, 2019-04-19:
Thank you for your work, I tested the new features, can work normally, the magic is the second mqtt client, I do not need to set the wifi ssid and password, it can work properly. This is incredible.

plapointe6, 2019-04-19:
Glad to hear that :)
The news functionalities will be included in the next release.

closed 2019-04-19 by plapointe6