plapointe6/EspMQTTClient #11: More sign changes?

ntableman, 2019-04-21:
So looks like there are more sig changes and I cannot pass in WiFi.macAddress() as the client name anymore. Any thoughts on how to make this work?

ntableman, 2019-04-21:
nm const char* mac = WiFi.macAddress().c_str();

plapointe6, 2019-04-21:
With recent updates, there are new signatures but it "should" not break codes already made. Can you send me how you declare the mqtt client and if possible the error you encounter ?

Thanks

ntableman, 2019-04-22:
So this used to work

`EspMQTTClient client(
  wifi_ssid,               // ssid
  wifi_pass,               // password
  "Punxbergbahn.lan",         // MQTT ip
  1883,                    // MQTT broker port
  "",                      // MQTT username
  "",                      // MQTT password
  WiFi.macAddress(),       // Client name
  onConnectionEstablished, // Connection established callback
  true,                    // Enable web updater
  true                     // Enable debug messages
);
`

and now I changed it to this per your notes

`EspMQTTClient client(
  wifi_ssid,                                      // ssid
  wifi_pass,                                      // password
  "Punxbergbahn.lan",  // MQTT ip
  "",                                             // MQTT username
  "",                                             // MQTT password
  ??????                                           // Client name
);
`

So I can no longer pass in the client as the Mac like I did, and to be honest I am trying to fix a few other  strange things that are maybe related:

So I have a setup with many subnets, and these devices are on one subnet, so the auto reg url in the updater doesn't work as they are not "local" to my computers. So I have been trying to set WiFi.hostname(myHostname) to the same thing in your "client name" so I can help keep things simple, but I cannot because of the processing order. Seems like in your code as I look at it, I can add a few lines that takes client name and make it WiFi.hostname(myHostname) as well. But the next thing I cannot seem to figure out is how to have the web updater URL it reports be the real domain name my DHCP server gives...

I suppose I should fix this all and have you decide if you like it and merge it in...

ntableman, 2019-04-22:
> void EspMQTTClient::connectToWifi()
> {
>   WiFi.mode(WIFI_STA);
>   **WiFi.hostname(mMqttClientName);**
>   WiFi.begin(mWifiSsid, mWifiPassword);
> 
> 

I made the above bold change in the .cpp file and now when I pass in the client name, it makes the hostname and client name the same, now to be able to make it more dynamic...

ntableman, 2019-04-22:
Now trying to figure out why http://B2:A0:50-switch.local. never works for the url of the web updater but the ip works.

plapointe6, 2019-04-22:
Have you tried this ? 
```c++
EspMQTTClient client(
  "WifiSSID",
  "WifiPassword",
  "192.168.1.100",  // MQTT Broker server ip
  WiFi.macAddress().c_str()     // Client name that uniquely identify your device
);
```

For the url of the web updater, I think that, since the v2.5.0 ESP8266 core, we need to call `MDNS.update();` in the `loop()` section. I will try it and if it solves the issue I will update the ESPMQTTClient.

Also, thanks for the clue about `WiFi.hostname(mMqttClientName);`, I think that is a good idea to add your proposed changes into the client, I will do it at the same time of adding `MDNS.update();`

plapointe6, 2019-04-22:
Ok, adding `MDNS.update()` in the `loop()`solve the issue with the web updater URL.
Also, I like the `WiFi.hostname(mMqttClientName)` change, I added that to the client.

I am waiting your feedback about the client name setting before releashing the changes above.

ntableman, 2019-04-25:
Sorry...day job :)

YUP! MDNS.update() in the loop works perfectly!

Also, this is kind of related: but I am a bit stumped, I want to change a bunch of variables including the topic and client I pass to  EspMQTTClient  based upon the MAC or some value I save in a JSON file on a SPIFF, but I cannot figure out how to execute a read of a SPIFF file or execute a bunch of IF's around  the MAC address before I run the EspMQTTClient constructor. Any ideas?

plapointe6, 2019-04-25:
You can use a pointer like this : 

```c++
EspMQTTClient* client;

setup()
{
  client = new EspMQTTClient(
    "WifiSSID",
    "WifiPassword",
    "192.168.1.100",  // MQTT Broker server ip
    WiFi.macAddress().c_str()     // Client name that uniquely identify your device 
  )
{

...
```

plapointe6, 2019-04-25:
Take note that you will need to use arrow instead of dot for calling functions of the client (`client->publish();`)

ntableman, 2019-04-25:
OK and now this one :)

So since I have 4 VLANs at home with 4 SSIDs for WiFi (I highly segregate personal devices from IoT, etc...) all these IoT thingies are on one VLAN/SSID and .local never seems to work as they are on HOST.iot.gsl.lan and my laptop is on HOST.gsl.lan so I was trying to figure out if I can change the way these things register with mDNS, etc... so the right url is used - but I admit I am likely an edge case.

plapointe6, 2019-04-25:
I admit that your question is a bit out of my knowledge ...
However, since we now set Wifi.hostname in the client, I think that, with the right configuration in your router, you will be able to reach hostname.iot.gsl.lan independently of the mDNS setup of the ESP ?
I have seen someone doing something similar but I dont exactly know how to do it. Something like running a DNS server in your router to substitute the IP adress of your ESP by HOSTNAME.iot.gsl.lan.

ntableman, 2019-04-25:
Yeah...it is a minor item...and I can make it work by just changing the text myself, which is not hard!

Thanks for the help!!!!

BTW I have a method that run every 5 minutes in loop() to send out an MQTT message to a topic which allows me to watch the status of my devices, I put MDNS.update(); in there and it also worked.

closed 2019-04-25 by plapointe6

ntableman, 2019-04-25:
Never mind the above the issue is that in my code a sneaky ":" got into the host name, so I found that and removed it and now all the .local's work!

plapointe6, 2019-04-25:
Glad to hear that :)