plapointe6/EspMQTTClient #12: RFE, please support wildcards (i.e., # and +)

atrent, 2019-05-14:
Hi, since the underlying PubSubClient (by knolleary) does support wildcards could it be possible to support them also in this lib?
Currently if I subscribe to a wildcard topic such as "root/sub/#" the subscription succeedes but no callback is called.
Thank you

res2cpu, 2019-05-27:
Good Evening

I though i was going mad on this one as i was having this same problem.

Would it be possible to add this?
Also what would be needed at that point would be a way to know what the topic was it came from, eg a second payload containing the topic (or the same payload containing the topic so it cab be split).

plapointe6, 2019-07-03:
Hello,

It should be possible and it can be a great improvement of this library.
I will take a closer look into this when I will get more time.

Regards

dralisz82, 2019-12-12:
Hi all,

I've just implemented a solution on this. See my fork of the lib: https://github.com/dralisz82/EspMQTTClient

My solution supports a single occurrence of a wildcard (+ or #), when subscribing, and extends the callback prototype with a 'topic' argument, so you must adapt all your callbacks as follows:
`	client.subscribe("topic", [](const String & topic, const String & payload) {
		Serial.println(topic + ": " + payload);
	});
`
See details in the code. :) and in the updated README and examples of course. ;)
I've also created a pull request, so feel free to use my solution.

BR,
Szilárd

plapointe6, 2020-02-09:
The wilcards support has been releashed with version 1.8.0

closed 2020-02-09 by plapointe6