plapointe6/EspMQTTClient #13: SimpleMQTTClient example won't compile

githubbuh, 2019-06-07:
plapointe6,

I found your project today, installed version 1.5.0 of the library using the Arduino library manager, and have tried to run the simple example.

Here are the compiler messages.

Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU-32S, 40MHz, 921600"

Build options changed, rebuilding all
In file included from /Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.h:22:0,
                 from /Users/macbookpro/Documents/Arduino/EspMQTTClient-master/examples/SimpleMQTTClient/SimpleMQTTClient.ino:8:
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:31:46: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   void setup(WebServer* server, char* path = "/", char* username = "", char* password = "")
                                              ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:31:68: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   void setup(WebServer* server, char* path = "/", char* username = "", char* password = "")
                                                                    ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:31:89: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   void setup(WebServer* server, char* path = "/", char* username = "", char* password = "")
                                                                                         ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h: In constructor 'ESP32HTTPUpdateServer::ESP32HTTPUpdateServer(bool)':
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:27:15: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     _username = "";
               ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:28:15: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     _password = "";
               ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h: In lambda function:
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:40:24: error: comparison with string literal results in unspecified behaviour [-Werror=address]
       if (_username != "" && _password != "" && !_server->authenticate(_username, _password))
                        ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:40:43: error: comparison with string literal results in unspecified behaviour [-Werror=address]
       if (_username != "" && _password != "" && !_server->authenticate(_username, _password))
                                           ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h: In lambda function:
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:60:26: error: comparison with string literal results in unspecified behaviour [-Werror=address]
         if (_username == "" || _password == "" || _server->authenticate(_username, _password))
                          ^
/Users/macbookpro/Documents/Arduino/libraries/EspMQTTClient/src/ESP32HTTPUpdateServer.h:60:45: error: comparison with string literal results in unspecified behaviour [-Werror=address]
         if (_username == "" || _password == "" || _server->authenticate(_username, _password))
                                             ^
cc1plus: some warnings being treated as errors
Multiple libraries were found for "WiFiClient.h"
 Used: /Users/macbookpro/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board NodeMCU-32S.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

alxstpl, 2019-06-14:
I have the same problem. Compilation for ESP32 failed, for ESP8266 ok and works.

githubbuh, 2019-06-14:
alxstpl,
What I have found is that if you change the compiler warning back to Default, the code runs despite the errors.
hope this helps.

alxstpl, 2019-06-14:
Thanks! But now i switched back to pubsubclient without this wrapper.

plapointe6, 2019-07-03:
I haven't ran into these errors when I tested with an ESP32 after adding support for it.
Thanks for pointing on this, I will take a closer look to that.

plapointe6, 2019-07-03:
Ok, I made some dumb errors in ESP32HTTPUpdateServer.h.
This can happen after too much coffee :unamused:
I will fix that really soon.

plapointe6, 2019-07-05:
These warnings are fixed in my side with the last commit (referenced above).
Do not hesitate to reopen this issue if it's not solved for you.

closed 2019-07-05 by plapointe6