plapointe6/EspMQTTClient #14: Validating that onConnectionEstablished() handles incoming messages

githubbuh, 2019-06-08:
plapointe6,

Do I understand correctly that onConnectionEstablished() is the callback for incoming mqtt messages?

If it is the case, I wonder why onConnectionEstablished() subscribes everytime rather than having the subscription in setup().

githubbuh, 2019-06-09:
plapointe6.

I understand now that onConnectionEstablished() is only called once after successfully connecting to both wifi and mqtt.

I think it would be helpful to modify the example to illustrate how ::subscribe(topic, pointer to function) may be used to specify a function that will process an incoming messages.

It might also be helpful to expose a class property that can be used, for example in onConnectionEstablished, to know if the call is at system startup (i.e. the first time), or because of a reconnect.

Maybe modify the simple example as follows:
`
// ------------------------------------------------------
// function to process incoming mqtt payload
//
// note that this function is connected to the subscriber
// in onConnectionEstablished()
//
// it is in this function that all of the processing of the 
// incoming MQTT payload will be performed
// ------------------------------------------------------
void topicPayloadProcess(String payload)
{
  Serial.println("in topicPayloadProcess"); 
  Serial.println(payload);
  mqttClient.publish("mytest/topic", payload);
}

// onConnectionEstablished() is called _**just once**_ after both 
// the WiFi client and the MQTT are successfully connected
//
// when a reconnect happens, because either the WiFi or MQTT
// client disconnects, onConnectionEstablished() is called again
//
// it is in this function that the MQTT subscriber is initialized
//
// when the subscriber is initialized, it is possible to 
// assign a function to process the topic payload
//
// NOTE : If this function is not implemented, then no subscriber
//        will be initialized, which means that the sketch will NOT
//        process incoming MQTT messages
void onConnectionEstablished()
{   
  // Initilize the subscriber to the topic and provide a function to
  // process the payload
  mqttClient.subscribe("mytopic/test", [](const String & payload) 
  {
      topicPayloadProcess(payload);
  });

  // an example showing that a message can be published from
  // onConnectionEstablished to notify that all is good.
  //
  // Publish a message to topic  (i.e. "mytopic/test")
  // Activate the MQTT retain flag by adding a third parameter of true

  // this is where it would be helpful to know if this is the first time or a reconnect
  // **NOTE to readers** of this comment in _this post_.  The IF statement shown
  // below is illustration only.  There is no reConnect() in the class today.
  if(mqttClient.reConnect())
  {
    mqttClient.publish("mytopic/text",
                                   "onConnectionEstablished() after reconnect"); 
  }
  else
  {
    mqttClient.publish("mytopic/text",
                                   "onConnectionEstablished() during system startup"); 
  }

  // An example of how to execute a delayed instruction, which will
  // delay processing a task without impact to 8266 nodes
  mqttClient.executeDelayed(5 * 1000, []() 
  {
    mqttClient.publish("mtopic/test", "This is a message sent 5 seconds later");
  });
}`

Thanks for the work that you have done to provide this library.  I have been struggling for several weeks now to come up with a sketch that has an availability of better than say 65%.  I haven't figured out yet why mine won't consistently reconnect.   

I am hoping that the EspMQTTClient will be just the ticket!

plapointe6, 2019-07-03:
Hi,
I like the idea of adding the possibility of knowing if the onConnectionEstablished was called after a connection lost or not. I will take note and maybe add this to the next version.

githubbuh, 2019-07-03:
plapointe6,

I have found two changes to be very helpful.
1) make serialLogsEnabled public.  It is a capability that can be used anywhere in the sketch.  Justification for exposing: since the sketch is focused on MQTT, makes sense to have access to useful features of the MQTT class.

2) add int mMqttConnectCount and int mqttConnectCount().  if == 1, then original connect.  if > 1 then reconnect.  This is very useful.  In the ping response, my nodes include connect count.  As illustrated below, the node lost its connection and reconnected at 17:34.
`
2019.6.30 15:00:02: {"mode":"diagnostic","event":"ping","sys":"esp32-irr","topic":"irrigation/monitor","action":"notify","msgID":"68f72347.c0178c","version":"2.0.1","ssid":"Hairy Feet","mac":"pending","mclient":"esp32-irr","free":274560,"invalid":0,"connect":1}
2019.6.30 17:34:09: {"mode":"diagnostic","event":"reconnect","sys":"esp32-irr","topic":"irrigation/monitor","action":"notify","version":"2.0.1","ssid":"Hairy Feet","mac":"pending","mclient":"esp32-irr","free":273400,"invalid":0,"connect":2}
`
The modified header.  Please note that this is a down-1 version.

`#ifndef ESP_MQTT_CLIENT_H
#define ESP_MQTT_CLIENT_H

#include <PubSubClient.h>

#ifdef ESP8266

  #include <ESP8266WiFi.h>
  #include <ESP8266WebServer.h>
  #include <ESP8266mDNS.h>
  #include <ESP8266HTTPUpdateServer.h>
  
  #define WebServer ESP8266WebServer
  #define ESPmDNS ESP8266mDNS
  #define ESPHTTPUpdateServer ESP8266HTTPUpdateServer

#else // for ESP32

  #include <WiFiClient.h>
  #include <WebServer.h>
  #include <ESPmDNS.h>
  #include "ESP32HTTPUpdateServer.h"

  #define ESPHTTPUpdateServer ESP32HTTPUpdateServer

#endif

#define MAX_TOPIC_SUBSCRIPTION_LIST_SIZE 10
#define MAX_DELAYED_EXECUTION_LIST_SIZE 10
#define CONNECTION_RETRY_DELAY 10 * 1000

void onConnectionEstablished(); // MUST be implemented in your sketch. Called once everythings is connected (Wifi, mqtt).

typedef void(*ConnectionEstablishedCallback) ();
typedef void(*MessageReceivedCallback) (const String &message);
typedef void(*DelayedExecutionCallback) ();

class EspMQTTClient 
{
private:
  // Wifi related
  bool mWifiConnected;
  unsigned long mLastWifiConnectionAttemptMillis;
  unsigned long mLastWifiConnectionSuccessMillis;
  const char* mWifiSsid;
  const char* mWifiPassword;
  WiFiClient mWifiClient;

  // MQTT related
  bool mMqttConnected;
  unsigned long mLastMqttConnectionMillis;
  const char* mMqttServerIp;
  const short mMqttServerPort;
  const char* mMqttUsername;
  const char* mMqttPassword;
  const char* mMqttClientName;
  bool mMqttCleanSession;
  char* mMqttLastWillTopic;
[... 622 lines cut ...]
  {
    strTerminationPos = length - 1;

    if (mEnableSerialLogs)
      Serial.print("MQTT! Your message may be truncated, please change MQTT_MAX_PACKET_SIZE of PubSubClient.h to a higher value.\n");
  }
  else
    strTerminationPos = length;

  // Second, we add the string termination code at the end of the payload and we convert it to a String object
  payload[strTerminationPos] = '\0';
  String payloadStr((char*)payload);

  // Logging
  if (mEnableSerialLogs)
    Serial.printf("MQTT >> [%s] %s\n", topic, payloadStr.c_str());

  // Send the message to subscribers
  for (int i = 0 ; i < mTopicSubscriptionListSize ; i++)
  {
    if (mTopicSubscriptionList[i].topic.equals(topic))
      (*mTopicSubscriptionList[i].callback)(payloadStr); // Call the callback
  }
}
`

plapointe6, 2019-07-03:
Please use the markdown syntax for formatting code when you copy/paste it on github.

I like your propositions. If you need to know id the Serial logs are enabled or not, it's better to add an accessor the the class (there is already a setter).

I also like your proposition of adding a member "mqttConnectCount".

It would be easier for me to review your code if you submit a pull request of the proposed changes instead of putting the whole code in this issue.,

Thanks for you contribution.

githubbuh, 2019-07-03:
As you have suggested, i will submit a pull request.

plapointe6, 2019-10-21:
This feature has been added to the 1.7.0 release.

closed 2019-10-21 by plapointe6