plapointe6/EspMQTTClient #18: How to use this library with a wired connection

zorce, 2019-07-15:
Hi,
I currently try to get this working on an ESP32-POE from Olimex. But noticed that it only supports wifi enabled connections. Even if you put the non managed connection without SSID. It still check for wifi.status()

Is it possible to enhance this to support for wired connection aswell?

Best regards!

plapointe6, 2019-07-15:
Hi,

I was not aware of the fact that there is some ESP32 wired board.

Yes, this library check wifi.status() even if you put the non managed connection without SSID. The reason for that is to handle MQTT things only when we are connected to a network.

I see two possibilities to make it work with your board : 
1. Modify the code to always handle MQTT without checking the status of the network connection when we dont set any SSID. This option must be tested as I am not sure that it will work well.
2. Find a way to detect when there is a wired connection and replace Wifi.status() with the right function for wired connection. Maybe you already have some clues on that ?

zorce, 2019-07-16:
Hi,
Thanks for the quick response. Managed to get it running using the
PubSubClient directly.
Not sure how I would pass the network reference to the library but will
manage implementing similar reconnection in the application code itself.

Best regards

closed 2019-07-17 by plapointe6