plapointe6/EspMQTTClient #19: Web updater not working

carranzafp, 2019-07-17:
Hello I have implemented a sketch based on the example

I enabled the HttpWeb Updater,  I went to http://mqttclient.local and I enter the mqtt user and password and the screen to upload the binary appears.

I choose the binary click on update and it says that it succeed and will reboot, but after reboot the behavior is the same as if the binary didnt change at all

If I take the device and upload the code via serial the change gets reflected.

```
EspMQTTClient client(
  "MySSID",
  "MyPWD",
  "192.168.x.xx",  // MQTT Broker server ip
  "user",   // Can be omitted if not needed
  "password",   // Can be omitted if not needed
  "mqttclient",      // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted
);

void setup()
{
   
  client.enableHTTPWebUpdater(); // Enable the web updater. User and password default to values of MQTTUsername and MQTTPassword. 
}

void loop()
{
  client.loop();
}
```

plapointe6, 2019-07-17:
Hello,

What board are you using and what is your version of EspMQTTClient ?

Regards

carranzafp, 2019-07-24:
Im using the ESP32 Doit
https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html

I originally raised this while I was on EspMQTTClient v1.5 but I just tested with 1.6 and still the same.  It seems not writting to the flash (and not resetting either, even when the web message says so)

Best Regards

carranzafp, 2019-07-24:
Hello  

I tried to run this example:
https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/

Its basically using the same libraries (WebServer, Update, etc) and it works perfectly, also the web GUI its improved.  

Here you have the log of the sucessful flash just in case that it helps:

```
Connected to MySSID
IP address: 192.168.1.205
mDNS responder started
Update: fast_blinker_test_testwebupdater.ino.doitESP32devkitV1.bin
Update Success: 763952
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

...
Connected to MySSID
IP address: 192.168.1.205
mDNS responder started

```

I could just implement that example but I love the idea behind of EspMQTTClient of having a single compact library, so I prefer to help you fix this (even when I can only help you testing)

plapointe6, 2019-08-10:
Hi,

Thanks for the help !
It should now be resolved with the 1.6.1 release.
Do not hesitate to re-open the issue if not.

Have a great day.

sixtyfive, 2019-08-30:
Unfortunately the Arduino IDE Libraries dialog still has 1.6.0 only :-/.

plapointe6, 2019-08-30:
Kind of odd ... I think I need to release a new version with the library.proprieties that match the version. I will do that right now.

sixtyfive, 2019-08-30:
Unfortunately now it crashes and never recovers (needing to be re-flashed over USB).

sixtyfive, 2019-08-30:
Let me know if there's any info I can provide...

plapointe6, 2019-08-30:
The web updater work for me with a LOLIN D32 Pro.

If you see a crash dump in the serial output, can you decode it with EspExceptionDecoder and send me the result ? (https://github.com/me-no-dev/EspExceptionDecoder).

Also, to be able to see if it's your board or the code, I can test your sketch with my board if you are willing to send it to me.

I find really odd the fact that you are able to run the exemple "https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/", but it crashes with ESPMQTTClient.

sixtyfive, 2019-08-31:
Okay, so it's a little weird: sometimes it'll work (but not be able to get wifi access after rebooting) and sometimes it won't. Unfortunately, whent it doesn't work at all, there's no stack trace to feed into the exception decoder, there's just this:

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

WiFi: Connecting to Zuhause ... 
WiFi: Connected, ip : 10.0.0.189
WEB: Updater ready, open http://Orno.local in your browser and login with username 'admin' and password 'orno'.
MQTT: Connecting to broker @10.0.0.1 ... ok.
MQTT: Subscribed to [Orno/pulses]
MQTT << [Orno/meta] connected
MQTT >> [Orno/pulses] 1082
Old pulse count: 1082
Old kWh count: 1.082kWh
Update: pulsecounter.ino.esp32.bin
No Error
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014f334: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IlleGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (
```

My sketch is basically still the same as over in the other Issue. Here's the link: https://gist.github.com/sixtyfive/4655787158b53685f4d5955e41944396.

plapointe6, 2019-09-04:
Do you ensure that the board wont be interrupted while doing the update process ? I see some attachInterrupt in your code. Regards.

sixtyfive, 2019-09-05:
I'll try disabling the interrupt handler before starting an update. My feeling is that that won't help with the WiFi not connecting after reboot. I'll report back!

sixtyfive, 2019-09-10:
Okay, finally had time to look at it again and ended up in a cul-de-sac ... there seems to be no way for `EspMQTTClient` of getting notified about the user [hitting the Upload button]([ESP32HTTPUpdateServer.h](https://github.com/plapointe6/EspMQTTClient/blob/master/src/ESP32HTTPUpdateServer.h#L48)). If there is, could you give an example?

plapointe6, 2019-10-02:
You could comment the interrupt part of your sketch to test if your issue come from this part.

I agree with you that it will probably won't help with the WiFi not connecting after reboot, maybe it's a completly different issue. Things are going odd with your ESP board.

plapointe6, 2020-02-09:
The original issue from @carranzafp has been resolved. The issue encontered by @sixtyfive is probably due to his cheap ESP clone or user error. Please open a new issue with details of how to reproduce the problem if it still happenning. Regards.

closed 2020-02-09 by plapointe6