plapointe6/EspMQTTClient #20: attachInterrupt Error

zsq19958629, 2019-07-23:
I used this code with the old version, "attachInterrupt(digitalPinToInterrupt(Int_Pin), IntCallback, RISING);" cars only Compile and run fine.
However, after the latest update, the compiled code caused repeated restarts on my esp8266.
`attachInterrupt(digitalPinToInterrupt(Int_Pin), IntCallback, RISING);

void IntCallback()
{
  KeyInt = 0;
  attachInterrupt(digitalPinToInterrupt(Key_Pin), KeyCallback, RISING);
}

void KeyCallback()
{
  KeyInt += 1; //Another program handles the KeyInt variable
}
`

Please help me. thank you

plapointe6, 2019-08-10:
Hello,

Before I take a closer look to this, I need to know
1- If you return to the previous version of EspMQTTClient, does it work again ? If not, maybe you updated the ESP8266 core at the same time of updating EspMQTTClient and the issue is not linked to EspMQTTClient.
2- If the first test show us that the issue appeared between two versions of EspMQTTClient, I need to know at wich version of EspMQTTClient it work and at wich version it wont work. This will help me figure out where it come from.
3- If the issue come from EspMQTTClient, I would need more details to reproduce the issue in my side. Your complete sketch would be great.

Regards

sixtyfive, 2019-08-30:
Not sure if I'm experiencing the same problem, but it does sound like it.
Whenever I call `client.publish()` from an interrupt handler, my ESP32 crashes and burns:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008b8c2  PS      : 0x00060f34  A0      : 0x8008aa0f  A1      : 0x3ffbe520  
A2      : 0x3ffc8a5c  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060f23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060f21  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x4010ee04  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008b8c2

Backtrace: 0x4008b8c2:0x3ffbe520 0x4008aa0c:0x3ffbe540 0x40089497:0x3ffbe560 0x4010ed81:0x3ffbe5a0 0x400fdb73:0x3ffbe5c0 0x4010f2b9:0x3ffbe5f0 0x4010fefa:0x3ffbe610 0x400fcbd0:0x3ffbe650 0x400fd3b5:0x3ffbe680 0x400d31dd:0x3ffbe6a0 0x400d28a9:0x3ffbe6e0 0x400d2b2d:0x3ffbe700 0x400d2b4d:0x3ffbe720 0x400d2250:0x3ffbe740 0x400d1c49:0x3ffbe760 0x40080f85:0x3ffbe7a0 0x40081789:0x3ffbe7c0 0x4000bfed:0x3ffb1dd0 0x4008a005:0x3ffb1de0 0x400890cd:0x3ffb1e00 0x4010ecc2:0x3ffb1e40 0x400fd464:0x3ffb1e60 0x400fd4f2:0x3ffb1e90 0x400d3289:0x3ffb1eb0 0x4014aaed:0x3ffb1ee0 0x400d2b5c:0x3ffb1f00 0x400d2449:0x3ffb1f50 0x400d1cba:0x3ffb1f90 0x400d7e2d:0x3ffb1fb0 0x4008877d:0x3ffb1fd0

Core 0 register dump:
PC      : 0x40089e93  PS      : 0x00060034  A0      : 0x8008b06d  A1      : 0x3ffb3c20  
A2      : 0x3ffbec34  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffb3c30  
A10     : 0x3ffc8bec  A11     : 0x3ffc8bec  A12     : 0x00060820  A13     : 0x00001980  
A14     : 0x00000000  A15     : 0x00000031  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x40089e93:0x3ffb3c20 0x4008b06a:0x3ffb3c50 0x400893f7:0x3ffb3c70 0x4010ecaa:0x3ffb3cb0 0x4010ef86:0x3ffb3cd0 0x400ff01c:0x3ffb3cf0 0x400ff085:0x3ffb3d10 0x4010431d:0x3ffb3d30 0x401049a0:0x3ffb3d60 0x4011069e:0x3ffb3dd0 0x401113b2:0x3ffb3df0 0x400fd9cd:0x3ffb3e10 0x4008877d:0x3ffb3e40

Rebooting...
```

My sketch looks like this:

```cpp
#include "config.h"

#include <EspMQTTClient.h>

EspMQTTClient client(
  WIFI_SSID, WIFI_PASSWORD,
  MQTT_SERVER,
  NULL, NULL,
  "Orno", 1883
);

long pulseCount = 0;
unsigned long pulseTime, lastTime;
double power, elapsedkWh;
const int ppwh = 1;
char mqtt_msg[12];

void onConnectionEstablished() {}

void setup()
{
  Serial.begin(115200);
  pinMode(PULSE_PIN, INPUT);
  attachInterrupt(digitalPinToInterrupt(PULSE_PIN), onPulse, FALLING);
}

void loop() {
  client.loop();
}

void onPulse()
{
  lastTime = pulseTime;
  pulseTime = micros();
  pulseCount++;
  power = (3600000000.0 / (pulseTime - lastTime)) / ppwh;
  elapsedkWh = (1.0 * pulseCount / (ppwh * 1000));

  dtostrf(elapsedkWh, 5, 3, mqtt_msg);  
  client.publish(MQTT_DATA_TOPIC, mqtt_msg);
}
```

Also tried moving `attachInterrupt()` into `onConnectionEstablished()`, but the result is the same: as soon as it hits `client.publish()` inside the interrupt handler: boom!

plapointe6, 2019-09-04:
I tested with an ESP8266 and it works well.

However, with an ESP32, I run into the same issue: it crashes when I call publish in the interrupt routine.

The issue does not seem to come from EspMQTTClient and probably not from PubSubClient.

The ESP32 seem to dislike being asked to send a TCP paquet while it is interrupted ...
I think that it's the kind of issue that needs to be addressed to the ESP32 core repo.

sixtyfive, 2019-09-05:
Interesting. I wonder if a simple workaround would be to use the second core when running on ESP32. Might work or might not...

plapointe6, 2019-09-05:
You can try this workaround until it's resolved (I cannot test it right now) : 
```c++
#include "config.h"

#include <EspMQTTClient.h>

EspMQTTClient client(
  WIFI_SSID, WIFI_PASSWORD,
  MQTT_SERVER,
  NULL, NULL,
  "Orno", 1883
);

long pulseCount = 0;
unsigned long pulseTime, lastTime;
double power, elapsedkWh;
const int ppwh = 1;
char mqtt_msg[12];

void onConnectionEstablished() {
  digitalWrite(LED_BUILTIN, HIGH);
}

void setup()
{
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(PULSE_PIN, INPUT);
  attachInterrupt(digitalPinToInterrupt(PULSE_PIN), onPulse, FALLING);
}

void loop() {
  client.loop();
}

void onPulse()
{
  lastTime = pulseTime;
  pulseTime = micros();
  pulseCount++;
  power = (3600000000.0 / (pulseTime - lastTime)) / ppwh;
  elapsedkWh = (1.0 * pulseCount / (ppwh * 1000));

  dtostrf(elapsedkWh, 5, 3, mqtt_msg);
  Serial.println(mqtt_msg);

  // This will be executed at the next client.loop() call
  client.executeDelayed(0, []() {
    client.publish(MQTT_DATA_TOPIC, mqtt_msg);
  });
}
```

zsq19958629, 2019-09-05:
fix it:
`//Before we add SETUP()
void ICACHE_RAM_ATTR IntCallback();
void ICACHE_RAM_ATTR KeyCallback();`

closed 2019-09-05 by zsq19958629

plapointe6, 2019-09-05:
Interresting !

So, you solved your random crashes on your ESP8266 by adding ICACHE_RAM_ATTR macro to your callback function like this ?

```c++
void ICACHE_RAM_ATTR onInterrupt();

setup(){
...
attachInterrupt(digitalPinToInterrupt(PIN), onInterrupt, RISING);
...
}

loop(){
...
}

void ICACHE_RAM_ATTR onInterrupt(){
// Interrupt routine
}
```

zsq19958629, 2019-09-05:
> Interresting !
>
> So, you solved your random crashes on your ESP8266 by adding ICACHE_RAM_ATTR macro to your callback function like this ?
>
> ```c++
> void ICACHE_RAM_ATTR onInterrupt();
>
> setup(){
> ... [...]

Yes, it works by adding it like this.

igorelterrible, 2019-09-24:
i had the same issue & it worked fine with the 
void ICACHE_RAM_ATTR onInterrupt();
......