plapointe6/EspMQTTClient #26: EspMQTTClient dynamic configuration

rumvit, 2019-09-17:
Hey. I have a good start of your class EspMQTTClient on the esp8266 platform .But I would like to have to adjust the connection settings after reading the SD card.Please tell me how can I change the connection settings to WIFI and MQTT broker in the code of the setup function?

plapointe6, 2019-09-17:
You could use a pointer to a EspMQTTClient and assign it in the setup() function with new.

```c++
EspMQTTClient* client;

setup() {
client = new EspMQTTClient(here fill your parameters as usual);
}
```

rumvit, 2019-09-17:
Thanks a lot.Everything seems to be obtained as wanted. Very good job.

closed 2019-09-17 by plapointe6

Sperryfreak01, 2023-04-01:
I get the following compile error when I attempt this solution.  Any recomendations?
_Request for member 'enableDebuggingMessages' in 'client', which is of pointer type 'EspMQTTClient*'_

My code is 
main.h
`EspMQTTClient* client;`

main.cpp
`client = new EspMQTTClient(
    WIFI_SSID, 
    WIFI_PASSWORD,
    MQTT_IP,  // MQTT Broker server ip
    MQTT_USER,   // Can be omitted if not needed
    MQTT_PASSWORD,   // Can be omitted if not needed
    mqttuser,     // Client name that uniquely identify your device  
    1883              // The MQTT port, default to 1883. this line can be omitted
  );

// Optional functionalities of EspMQTTClient
  client.enableDebuggingMessages(); // Enable debugging messages sent to serial output
  client.enableHTTPWebUpdater(); // Enable the web updater. User and password default to values of MQTTUsername and MQTTPassword. These can be overridden with enableHTTPWebUpdater("user", "password").
  client.enableOTA(); // Enable OTA (Over The Air) updates. Password defaults to MQTTPassword. Port is the default OTA port. Can be overridden with enableOTA("password", port).
  client.enableLastWillMessage(deviceChannel, "Disconnected");  // You can activate the retain flag by setting the third parameter to true
  `

Sperryfreak01, 2023-04-01:
I am only slightly less dumb than I look 😁 

For anyone else that comes this way, make sure to refer to it via `client->[method]` instead of client.[method] before you updated it to be dynamic.  I'm not great with pointers so took me a bit to understand the error message.