plapointe6/EspMQTTClient #32: How to support TLS？

KISSMonX, 2019-11-10:


plapointe6, 2019-11-12:
Unfortunatly, there is no SSL/TLS support yet with this lib. I will add that to the TODO list. Here is a PubSubClient issue discussing how you can achieve this with the PubSubClient lib : https://github.com/knolleary/pubsubclient/issues/462

KISSMonX, 2019-11-14:
Thank you for your reply, happy to look forward to the arrival of new features. 😊

steros76, 2019-11-14:
This feature would be very welcome!
I hope it's on first line of your todo list... :)

cstanke, 2019-12-04:
Yes please! Yes please!

nopap, 2020-04-06:
Hi,
Updating the library is in fact quite simple to make it work in TLS. 
It's probably quick and dirty, but works for me.

Change the include from
  `#include <WiFiClient.h>` to `#include <WiFiClientSecure.h>` 
 
Then the declaration:
  `WiFiClient mWifiClient;`  to  `WiFiClientSecure mWifiClient; `
and move it to public.

Then the library will use the secure library, and give you access from your code to the mWifiClient object where you'll be able to change settings and set properties such as:
  ```
 client.mWifiClient.setFingerprint(fingerprint);   // <-- Set the SHA1 fingerprint to verify remote server
    // Alternative: client.mWifiClient.setInsecure();
```

Works for my mosquitto in mqtts.

codersaur, 2020-04-26:
+1. Please add support for TLS (ideally server and client certs). Thanks.

paviro, 2021-01-17:
I would also really appreciate SSL support for [HASS-YAAP](https://github.com/paviro/HASS-YAAP) :)

cd334, 2022-12-30:
+1. Please add support for TLS! Thanks!

zouje, 2023-01-14:
+1 please

EdJoPaTo, 2023-01-14:
There is no need in adding "+1" comments. That does not change anything. You could instead provide Pull Requests. ;)

plapointe6, 2023-01-15:
If someone takes time to implement this, I will take the time to review and test. However, I don't think I will implement this by myself. I work on this lib for free and I don't have as many free time as before. Also, as almoust all of my home automation devices are now working on ZigBee, I don't use this lib as much as before. It became hard for me to test new functionnalities. Thank you for your comprehension.