plapointe6/EspMQTTClient #33: Autoreconnect feature

steros76, 2019-11-17:
If ESP lose Wifi signal, this lib should try to reconnect. Now connection is lost until I reset ESP.
I think this is very important for IOT devices.

plapointe6, 2019-11-17:
Thanks for reporting this, I will make this issue my priority.

abqmichael, 2019-11-18:
Hi. I just pinned down this issue myself when running ESP32 boards. I had noticed that my device was going zombie randomly. In tracking this down, I programmed 3 different ESP32 boards to run the same software and found that they all went zombie together. Looking on my network, I found that my router was misbehaving (a hard reset cured that). Moving back to the tests, I found that the clients would reconnect just fine if I unplugged my MQTT server for several minutes but do not appear to reliably reconnect if I unplugged my router. (I only tested that once.)

In parallel to the tests, I found an article describing the same problem reported related to PubSubClient, an underlying libraries used in EspMQTTClient. In their investigation, they found that the problem was more with the example code than PubSubClient itself. Apparently the connection code needed to test a variety of conditions. They concluded that software should connect very differently than in the example and they gave some example code for doing this. (https://github.com/knolleary/pubsubclient/issues/444).

I only barely started investigating whether this would be right for EspMQTTClient. Within the code, I see that the connect/reconnect software is written quite differently than the discussion. As a result, it might not be as simple to patch as I hoped. However, their guide may be useful just the same.

kkberg, 2019-11-29:
Hello, having also connection issue when collecting data only every 15 minutes and want to push to MQTT-broker. I then tried to use a loop instead of only delay() function. But after 2 connects in 2 minutes connection ist lost.
[image]
Thank you

plapointe6, 2019-11-29:
Have-you a ESP8266 or ESP32 ? I am pretty sure that the connection issue arise with ESP32 only. It will guide my troubleshooting.

kkberg, 2019-11-29:
I have an ESP8266 12E (WEMOS D1 Mini) it also happens on other NodeMCU's like the Lolin .

plapointe6, 2019-11-29:
@kkberg make sure that you never "block" the main core for too long, as it will result to a connection timeout. The reason is that there is only one core to handle the communication and to execute your code. Don't use the delay function. Exemple of what you could do : 
 ```c++

#include "EspMQTTClient.h"

EspMQTTClient client(
.....
);

void setup()
{
.....
}

void onConnectionEstablished()
{
  // Execute delayed instructions, here is a 5 seconds delay
  client.executeDelayed(5 * 1000, readSensor);
}

void readSensor() 
{
	// IMPORTANT, protect from spawning multiple delayed instructions when the connection is lost
	if(!client.isConnected())
		return

	// What you wanbt to execute each (5) seconds
	client.publish("mytopic/test", "This is a message sent with a delay of 5 seconds");
	
	// Re-schedule the instructions
	client.executeDelayed(5 * 1000, readSensor);
}

void loop()
{
  client.loop();
}
```

kkberg, 2019-11-29:
@plapointe6 

Thank you very much, I integrated your solution and I got it to work.

MoosHam, 2020-01-17:
Hi all,

I know the failing reconnect on lost Wifi is a bug and will be dealt with. But how do you guys work around this right now? I also suffer from it but don't know how to deal with it, other than jsut occasionally rebooting the ESP32 once I suspect it is hanging...

thanks,
MoosHam

abqmichael, 2020-01-17:
Sadly, I don't have a work-around. I have traced the problem to loosing WiFi. In fact, I can get it to reliably fail by just turning my WiFi router off and waiting a little.

I tried the suggestion about not sending messages during a callback and it didn't change anything for me. Frankly, it turned out to not be my problem. I decided to dig into the library code to see what I could find out. What I found is that after the WiFi gets lost, eventually EspMQTTClient::connectToMqttBroker() gets called (probably from from EspMQTTClient::loop()) and this, in turn, calls mMqttClient.connect (mMqttClient is a PubSubClient instance) and that method hangs.
 
Tracing down from EspMQTTClient::loop(), I see that connectToMqttBroker() gets called when loop() discovers that it has lost the MQTT client connection. Oddly, though loop() does discover that, it does not recognize at that time that it has lost WiFi. That is, it appears that WiFi.status() == WL_CONNECTED even though the WiFi is gone. 

Right now, I'm trying to see if I can fake things out so that the ESP32 forces a disconnect when it looses WiFi and then will re-establish then. However, my first stab at that idea failed so perhaps I'll have to dig further.

As an aside, I've read that other people are having problems keeping their ESP32s connected to specific routers. My router (an Apple Time capsule) is on that list of bad boys. I might punt and buy a different wifi router.

Michael
> 
> I know the failing reconnect on lost Wifi is a bug and will be dealt with. But how do you guys work around this right now? I also suffer from it but don't know how to deal with it, other than jsut occasionally rebooting the ESP32 once I suspect it is hanging...
> 
> thanks,
> MoosHam

MoosHam, 2020-01-20:
Hi Michael,

great to see you seem to be getting to the bottom of this issue. Quite annoying little bug. BTW, I use a ubiquity AP and have similiar issues. It's happening only from time to time (every few days or so), which makes it hard to catch...

MoosHam

abqmichael, 2020-01-20:
I’m getting confident that this can be done. I found some complicated code that used another MQTT library at https://esp32.com/viewtopic.php?t=3851 <https://esp32.com/viewtopic.php?t=3851>  I say complicated mainly because it also used the secure connection to MQTT and I haven’t gotten there in configuring my MQTT server. I modified this code and got it to work with my MQTT server and it survived the wifi re-boot.

I included my version below. Now I’m working to integrate the idea into the EspMQTTClient library. 

Michael

/* standalone device with MQTT function
    Code found at https://esp32.com/viewtopic.php?t=3851

   The requirements are as follows:
   - it should run its main tasks regardless of the availability of WiFi/MQTT
   - it should recover lost WiFi/MQTT without interrupting the main tasks
   - unless other examples on the net it does not start the WiFi/MQTT connection in the setup() section
     since the device should start with its main functions immediately and do the connection setup later

   Other features
   - OTA over WiFi using the Arduino IDE
   - MQTT over TLS for encrypted data transfer
   - configuration parameters for device name, SSID, WAP-PW, MQTT broker username and password are stored in Preferences

    Feedback to improve this code is welcome
    Urs Eppenberger
*/

#include <WiFi.h>
#include <MQTT.h>
// Shows up in Arduino Library as MQTT by Joel Gaehwiler

const char* Hostname = "esp32_wt";            // change according your setup : it is used in OTA and as MQTT identifier
String WiFi_SSID = "AbqMichaelJ";               // change according your setup : SSID and password for the WiFi network
String WiFi_PW = "BananaNutBread";                   //    "
const char* OTA_PW = "BananaNutBread";                // change according your setup : password for 'over the air sw update'
String mqtt_broker = "10.0.1.201";         // change according your setup : IP Adress or FQDN of your MQTT broker
String mqtt_user = "esp32_wt";               // change according your setup : username and password for authenticated broker access
String mqtt_pw = "";                   //    "
String input_topic = "wifi/test";        // change according your setup : MQTT topic for messages from device to broker
unsigned long waitCount = 0;                 // counter
uint8_t conn_stat = 0;                       // Connection status for WiFi and MQTT:
//
// status |   WiFi   |    MQTT
// -------+----------+------------
//      0 |   down   |    down
//      1 | starting |    down
//      2 |    up    |    down
//      3 |    up    |  starting
//      4 |    up    | finalising
//      5 |    up    |     up

unsigned long lastStatus = 0;                // counter in example code for conn_stat == 5
unsigned long lastTask = 0;                  // counter in example code for conn_stat <> 5

const char* Version = "{\"Version\":\"low_prio_wifi_v2\"}";
const char* Status = "{\"Message\":\"up and running\"}";

WiFiClient TCP;                        // TCP client object, uses SSL/TLS
MQTTClient mqttClient(512);                  // MQTT client object with a buffer size of 512 (depends on your message size)

void setup() {
  Serial.begin(115200);
  // configure mqttClient
}

String message = "None";

void loop() {                                                     // with current code runs roughly 400 times per second
  // start of non-blocking connection setup section
  if ((WiFi.status() != WL_CONNECTED) && (conn_stat != 1)) {
    conn_stat = 0;
  }
  if ((WiFi.status() == WL_CONNECTED) && !mqttClient.connected() && (conn_stat != 3))  {
[... 47 lines cut ...]
      conn_stat = 5;
      break;
  }
  // end of non-blocking connection setup section

  // start section with tasks where WiFi/MQTT is required
  if (conn_stat == 5) {
    if (millis() - lastStatus > 10000) {                            // Start send status every 10 sec (just as an example)
      Serial.println(Status);
      mqttClient.publish(input_topic, Status);                      //      send status to broker
      mqttClient.loop();                                            //      give control to MQTT to send message to broker
      lastStatus = millis();                                        //      remember time of last sent status message
    }
    mqttClient.loop();                                              // internal household function for MQTT
  }
  // end of section for tasks where WiFi/MQTT are required

  // start section for tasks which should run regardless of WiFi/MQTT
  if (millis() - lastTask > 1000) {                                 // Print message every second (just as an example)
    Serial.println(message);
    lastTask = millis();
  }
  delay(100);
  // end of section for tasks which should run regardless of WiFi/MQTT
}

abqmichael, 2020-01-21:
As to work-around, the first thing that helped was to hard reset my apple time capsule and bring my network down to just that. (I had some apple devices extending my network and I turned them off.) I'm guessing there must have been a problem as the disconnect frequency went way down. You might wish to have a look at your wifi access point.

Oh... I got my test code working with PubSubClient so now it's time to start modifying the EspMQTTClient library. I'm hoping I can do that this week. I will only be able to post that as a suggested fix and Patrick will need to analyze/test and accept or reject it.

plapointe6, 2020-01-21:
Great, I can help if you have questions. One of the challenge of this change will be to keep the ESP8266 compatibility.

abqmichael, 2020-01-22:
Thankfully the code I'm patching doesn't have any of the ESP8266 specific calls in it. You had nicely separated that into methods called by the patch code. I got that working but I am having a problem right now with the callback execution. I may have inadvertently broken something. So I'm going to go back to your latest version and more carefully push in my patch with attention to keeping the code around that piece. 

One question I have is whether the callbacks need the MQTT connected. I think that's how it was implemented. However, it would be nice if the logic in my applications could keep running whether the network was up or down. (My application would run fine off network. MQTT is only needed to tell it to do new things.)

plapointe6, 2020-01-22:
Not sure to understand your question, maybe some code can help. The callbacks are here to trigger when something is received from MQTT, so they won't be triggered if the MQTT is not connected. Maybe the problem you face is linked to #21, when the networks go down, the PubSubClient library blocks the sketch. It's an issue that I plan to resolve some day by implementing my own version of PubSubClient. Sometime there is workarounds to prevent that from happening, it depends on the cases.

abqmichael, 2020-01-22:
I am referring to executeDelayed. One of my favorite features of EspMQTTClient is the whole event system vis executeDelayed. For example, I use it to animate lights vis FastLED. I want my light animation party to go on when the WiFi goes down. Besides not going back online, that’s what happens in the current release.

abqmichael, 2020-01-24:
I have a version that's working and am now doing cleanup and optimization. How would you like me to get it to you?

plapointe6, 2020-01-24:
If you plan to do a pull request, it's better if you finish your cleanup and create a pull request. This way I will easily be able to see changes and comment parts of the code :)

abqmichael, 2020-01-24:
Hi Patrick;

I’m very novice at git and I don’t imagine that I have write access to your repository. (My brain just can’t parse git lingo. For example, I’ve no real idea what it means to "Generate a request asking your upstream project to pull changes into their tree.”)

If you will email me at abqmichaelj@aol.com <mailto:abqmichaelj@aol.com>, I can send you a copy of the code. Alternatively, I could put it in one of these comments. I suppose I could also do the git thing if you wish to make me a contributor to your project. 

To make my development and your seeing what I’ve done easier, I have demarked all my relevant changes vs the original between #ifdef WIFI_FIX/#else/#endif blocks. (This let me quickly go back and forth between versions.) I am working from the most recent version of the code that’s on GitHub.

plapointe6, 2020-01-24:
To do a pull request and become a contributor of this project, make a fork of this lib, commit your changes to your fork and then you will be able to do a pull request : https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

If you prefer to not do this procedure, I can get your code and adapt it by my own.

abqmichael, 2020-01-27:
#ifndef ESP_MQTT_CLIENT_H
#define ESP_MQTT_CLIENT_H

#include <PubSubClient.h>

#ifdef ESP8266

  #include <ESP8266WiFi.h>
  #include <ESP8266WebServer.h>
  #include <ESP8266mDNS.h>
  #include <ESP8266HTTPUpdateServer.h>
  
  #define WebServer ESP8266WebServer
  #define ESPmDNS ESP8266mDNS
  #define ESPHTTPUpdateServer ESP8266HTTPUpdateServer

#else // for ESP32

  #include <WiFiClient.h>
  #include <WebServer.h>
  #include <ESPmDNS.h>
  #include "ESP32HTTPUpdateServer.h"

  #define ESPHTTPUpdateServer ESP32HTTPUpdateServer

#endif

#define MAX_TOPIC_SUBSCRIPTION_LIST_SIZE 10
#define MAX_DELAYED_EXECUTION_LIST_SIZE 10
#define CONNECTION_RETRY_DELAY 10 * 1000

#define WIFI_FIX
#ifdef WIFI_FIX
#pragma message "Using NEW connection algorithm V 1/23/20 @ 10:45"
#endif

void onConnectionEstablished(); // MUST be implemented in your sketch. Called once everythings is connected (Wifi, mqtt).

typedef void(*ConnectionEstablishedCallback) ();
typedef void(*MessageReceivedCallback) (const String &message);
typedef void(*MessageReceivedCallbackWithTopic) (const String &topicStr, const String &message);
typedef void(*DelayedExecutionCallback) ();

class EspMQTTClient 
{
private:
#ifdef WIFI_FIX
  // mConnState values
  // status |   WiFi   |    MQTT
  // -------+----------+------------
  //      0 |   down   |    down
  //      1 | starting |    down
  //      2 |    up    |    down
  //      3 |    up    |  starting
  //      4 |    up    | finalising
  //      5 |    up    |     up
  uint8_t mConnState; // connection state (1-5)
  unsigned int mWaitCount;
  unsigned int mBrokerConnectPauseMills; // minimum time (mills) allowed between calls to mMqttClient.connect
  // Wifi related
#else
  bool mWifiConnected;
  unsigned long mLastWifiConnectionAttemptMillis;
  unsigned long mLastWifiConnectionSuccessMillis;
#endif
  const char* mWifiSsid;
  const char* mWifiPassword;
  WiFiClient mWifiClient;

  // MQTT related
[... 124 lines cut ...]
  void executeDelayed(const unsigned long delay, DelayedExecutionCallback callback);
#ifdef WIFI_FIX
  bool isConnected(); // Return true if everything is connected
  bool isWifiConnected() ; // Return true if wifi is connected
  bool isMqttConnected(); // Return true if mqtt is connected
#else
  inline bool isConnected() const { return isWifiConnected() && isMqttConnected(); }; // Return true if everything is connected
  inline bool isWifiConnected() const { return mWifiConnected; }; // Return true if wifi is connected
  inline bool isMqttConnected() const { return mMqttConnected; }; // Return true if mqtt is connected
#endif
  inline bool getConnectionEstablishedCount() const { return mConnectionEstablishedCount; }; // Return the number of time onConnectionEstablished has been called since the beginning.

  inline void setOnConnectionEstablishedCallback(ConnectionEstablishedCallback callback) { mConnectionEstablishedCallback = callback; }; // Default to onConnectionEstablished, you might want to override this for special cases like two MQTT connections in the same sketch

private:
  void connectToWifi();
  void connectToMqttBroker();
  bool mqttTopicMatch(const String &topic1, const String &topic2);
  void mqttMessageReceivedCallback(char* topic, byte* payload, unsigned int length);
#ifdef WIFI_FIX
  void configureHTTPWebUpdater(); // Worker method to configure web updater. 
#endif
};

#endif

abqmichael, 2020-01-27:
#include "EspMQTTClient.h"

// =============== Constructor / destructor ===================

// MQTT only (no wifi connection attempt)
EspMQTTClient::EspMQTTClient(
  const char* mqttServerIp,
  const short mqttServerPort,
  const char* mqttClientName) :
  EspMQTTClient(NULL, NULL, mqttServerIp, NULL, NULL, mqttClientName, mqttServerPort)
{
}
EspMQTTClient::EspMQTTClient(
  const char* mqttServerIp,
  const short mqttServerPort,
  const char* mqttUsername,
  const char* mqttPassword,
  const char* mqttClientName) :
  EspMQTTClient(NULL, NULL, mqttServerIp, mqttUsername, mqttPassword, mqttClientName, mqttServerPort)
{
}

// Wifi and MQTT handling
EspMQTTClient::EspMQTTClient(
  const char* wifiSsid,
  const char* wifiPassword,
  const char* mqttServerIp,
  const char* mqttClientName,
  const short mqttServerPort) :
  EspMQTTClient(wifiSsid, wifiPassword, mqttServerIp, NULL, NULL, mqttClientName, mqttServerPort)
{
}

// Warning : for old constructor support, this will be deleted soon or later
EspMQTTClient::EspMQTTClient(
  const char* wifiSsid, 
  const char* wifiPassword, 
  const char* mqttServerIp,
  const short mqttServerPort, 
  const char* mqttUsername, 
  const char* mqttPassword,
  const char* mqttClientName, 
  ConnectionEstablishedCallback connectionEstablishedCallback,
  const bool enableWebUpdater, 
  const bool enableSerialLogs) :
  EspMQTTClient(wifiSsid, wifiPassword, mqttServerIp, mqttUsername, mqttPassword, mqttClientName, mqttServerPort)
{
  if (enableWebUpdater)
    enableHTTPWebUpdater();

  if (enableSerialLogs)
    enableDebuggingMessages();

  setOnConnectionEstablishedCallback(connectionEstablishedCallback);

  mShowLegacyConstructorWarning = true;
}

// Warning : for old constructor support, this will be deleted soon or later
EspMQTTClient::EspMQTTClient(
  const char* wifiSsid, 
  const char* wifiPassword,
  ConnectionEstablishedCallback connectionEstablishedCallback, 
  const char* mqttServerIp, 
  const short mqttServerPort,
  const char* mqttUsername,
  const char* mqttPassword, 
  const char* mqttClientName,
  const bool enableWebUpdater,
  const bool enableSerialLogs) :
[... 644 lines cut ...]
  }
  else
    strTerminationPos = length;

  // Second, we add the string termination code at the end of the payload and we convert it to a String object
  payload[strTerminationPos] = '\0';
  String payloadStr((char*)payload);
  String topicStr(topic);

  // Logging
  if (mEnableSerialLogs)
    Serial.printf("MQTT >> [%s] %s\n", topic, payloadStr.c_str());

  // Send the message to subscribers
  for (byte i = 0 ; i < mTopicSubscriptionListSize ; i++)
  {
    if (mqttTopicMatch(mTopicSubscriptionList[i].topic, String(topic)))
    {
      if(mTopicSubscriptionList[i].callback != NULL)
        (*mTopicSubscriptionList[i].callback)(payloadStr); // Call the callback
      if(mTopicSubscriptionList[i].callbackWithTopic != NULL)
        (*mTopicSubscriptionList[i].callbackWithTopic)(topicStr, payloadStr); // Call the callback
    }
  }
}

abqmichael, 2020-01-27:
Hi 

> If you plan to do a pull request, it's better if you finish your cleanup and create a pull request. This way I will easily be able to see changes and comment parts of the code :)

I simply could not find my way through the pull request process. The documentation you suggested assumes that I've done something to prepare my files, like create a git repository of my own or something, but doesn't say what that might be. 

I tried clicking into pull request but it would not let me submit my source. I downloaded the GIT tool, then your repository, checked in a local copy but didn't have permission to upload that to your repository and didn't see how that could help me create something that GIT would recognize/use as a pull request. 

Sorry, but I'm going to have Scotty beam me out of git hell for now. My code is above. Would be happy to email you a copy too.

abqmichael, 2020-01-27:
Frustration resolved. I found a beginners guide to doing this at https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/ and was able to create a fork, push the data up to GitHub and then create a pull request. Kudos to the git developers and documenters for making this relatively simple thing so opaque.

abqmichael, 2020-01-28:
Here is a state diagram explaining the new connection algorithm.

[image]

plapointe6, 2020-02-09:
It will take time to review your changes and test them with an ESP32 and ESP8266, please be patient and thanks for your contribution.

abqmichael, 2020-02-10:
Thanks for letting me know. I have been using it on 3 devices since posting about it and it’s working great. Not one crash. I monitored one ESP32 via the serial port for several days and found that it would have crashed four times over that period without the changes. My only timing concern would be merging my changes back in if the library was updated without the changes.

Michael J.

ssp5zone, 2020-07-21:
Not sure if you gave a fix. But seem to be working now. 🥳 

Just tested it on ESP32-WROOM.

> 01:40:22.633 -> WiFi! Lost connection.
> 01:40:22.633 -> 
> 01:40:22.633 -> WiFi: Connecting to Wi5Zone ... 
> 01:40:41.851 -> WiFi: Connecting to Wi5Zone ... 
> 01:41:00.197 -> WiFi: Connecting to Wi5Zone ... 
> 01:41:22.332 -> WiFi: Connected, ip : 192.168.0.100
> 01:41:22.332 -> MQTT: Connecting to broker @io.adafruit.com ... ok.

Cheers! 🍻

plapointe6, 2020-09-03:
Version 1.9.0 introduce an overhaul of the connection mecanism to make the ESP32 connection a lot more stable. I will close this issue, but do not hesitate to re-open another issue more recent with updated details if there still ESP32 wifi connection issues. Regards.

closed 2020-09-03 by plapointe6