plapointe6/EspMQTTClient #36: Does EspMQTTClient support wemos D1 R1?

marryton007, 2019-12-24:
D1 is An Arduino UNO Compatible wifi board based on ESP8266EX, please see the [wiki](https://wiki.wemos.cc/products:d1:d1)

hixfield, 2019-12-24:
I have developed a few projects on the WeMos D1 R1 and this works perfectly for me.
Make sure to select the correct board in Arduino -> Settings -> Board = "WeMos D1 R1"

marryton007, 2019-12-24:
Thank your help!  I use vscode + platformio to develope a example project, I don't know why the serial console don't ouput anything, I have called the  function 'enableDebuggingMessages', it seems that it didn't connect the WIFI and MQTT server.

I'll try it on  the Arduino IDE.  thanks.

marryton007, 2019-12-25:
I have found the reason that  It can't include the "SPI.h",  but  I don't know which one cause this.

marryton007, 2019-12-25:
I found the reason, under the vscode + platformIO, it can't include the header file "SPI.h" , but I tried it in Arduino IDE,  it is OK. I'll do more test on Arduino IDE.

Bozzibo, 2020-01-03:
I have a compile error on the wemos D1 mini.
"exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini."

The mini is not supported? It is also ESP8266.

plapointe6, 2020-01-06:
Wemos D1 mini is supported, I use this lib mostly with then. You will notice the real cause earlier in the logs.

closed 2020-01-06 by plapointe6

Bozzibo, 2020-01-06:
> Wemos D1 mini is supported, I use this lib mostly with then. You will notice the real cause earlier in the logs.

Can't get it to compile. Which cause are you referring to? 
I am using ESP boards 2.6.3

plapointe6, 2020-01-06:
Can you send me the complete logs please ? Wich IDE are you using ?

Bozzibo, 2020-01-06:
Never mind, my fault. 
I was missing pubsubclient.. 
Thanks for the support ;)