plapointe6/EspMQTTClient #40: 'id' is not captured

StephanMeijer, 2020-01-15:
```
void onConnectionEstablished()
{
  for (Light l : lights) {
    String id = l.getId();

    client.subscribe("light/" + id + "/switch", [](const String & payload) {
      if (payload == "ON") {
        turnOn(id);
        client.publish("light/" + id + "/state", "ON");
      } else if (payload == "OFF") {
        turnOn(id);
        client.publish("light/" + id + "/state", "OFF");
      }
    });
  }
}
```

Probably a stupid question: how can I use variables and class instances from outside the callback in the callback itself?

plapointe6, 2020-01-16:
Hello,

If you define your variables and classes outside every functions (global scope), you will have access to them. It's a common practice with Arduino sketches.

See : https://www.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/scope/

plapointe6, 2020-01-16:
You could try to capture the scope, but it may not work because it will probably change the signature of the callback: 

```c++
void onConnectionEstablished()
{
...
    client.subscribe("light/" + id + "/switch", [&](const String & payload) {
...
```

if it does not work you will need to define your "String id" variable in the global scope (ugly workaround)

StephanMeijer, 2020-01-16:
I created a PR to make this solution work.

StephanMeijer, 2020-01-16:
That's what I thought yesterday after debugging, but I am not familiar with C++.

plapointe6, 2020-01-16:
Related to pr #41

plapointe6, 2020-02-09:
Capturing support has been released in version 1.8.0

closed 2020-02-09 by plapointe6