plapointe6/EspMQTTClient #42: Can I higher up MAX_TOPIC_SUBSCRIPTION_LIST_SIZE?

StephanMeijer, 2020-01-18:
```
#define MAX_TOPIC_SUBSCRIPTION_LIST_SIZE 10
```

MoosHam, 2020-01-20:
This is a great point, was pulling my hair out over this :-)

Is this the setting that is preventing me to subscribe to more than 10 topics?
Can I jsut change that?
Why is it set to 10, are there downsides when increasing to 20 for example?

thanks,
MoosHam

plapointe6, 2020-01-20:
@StephanMeijer @MoosHam, is it enough if I increase that to 20 ? I tought that 10 was enough for everyone, no the case so I will increase it.

StephanMeijer, 2020-01-20:
It would be better to be able to increase it from outside the library.

StephanMeijer, 2020-01-20:
Like maybe to give it as a property when the `Client`-object is being initiated.

plapointe6, 2020-01-20:
Yes, but this will lead to dynamic allocation of memory, I prefer to avoid that with microcontrollers when I can. However, maybe we are in a case where it's better to do so. I will think about that.

StephanMeijer, 2020-01-20:
I trust your expertise in this.
Is there any way we can perhaps wrap the definition on your side in a `IFDEF`, so we can define the value on our side? Or doesn't it work that way?

plapointe6, 2020-01-20:
Yes, it's a good idea, I will do so right away.

plapointe6, 2020-01-20:
In fact, I think that it is not necessary. Just do that after the EspMQTTClient.h include : 
```c++
#ifdef MAX_TOPIC_SUBSCRIPTION_LIST_SIZE
#undef MAX_TOPIC_SUBSCRIPTION_LIST_SIZE
#define MAX_TOPIC_SUBSCRIPTION_LIST_SIZE 20
#endif
```

closed 2020-02-09 by plapointe6

StephanMeijer, 2020-03-15:
[image]
[image]
Does not seem to work..