plapointe6/EspMQTTClient #45: mqtt disconnecting

skibum-za, 2020-02-12:
I did wonder if I was sufferring from the wifi disconnection issue reported. I was certainly doing somethings you say you shouldn't do (like using delay()).

So, I went back to basics and just used the sample code, but had the executeDelayed() call my function as you posted in the wifi disconnection thread.

While my wifi doesn't disconnect, the mqtt does. This happened only 16 minutes after starting and is very repeatable. Sometimes even quicker. 

```
14:26:28.869 -> Enterered my_function()
14:26:28.869 -> isConnected (1) - isWifiConnected (1) - isMqttConnected (1)
14:26:28.869 -> MQTT << [mytopic/test] This is a message
14:26:28.869 -> Going to sleep now
14:26:38.873 -> Enterered my_function()
14:26:38.873 -> isConnected (1) - isWifiConnected (1) - isMqttConnected (1)
14:26:38.873 -> MQTT << [mytopic/test] This is a message
14:26:38.873 -> Going to sleep now
14:26:38.873 -> MQTT! Lost connection.
14:26:38.873 -> MQTT: Connecting to broker @pi4-4zig ... unable to connect, MQTT_CONNECT_FAILED
14:26:48.870 -> Enterered my_function()
14:26:48.870 -> isConnected (0) - isWifiConnected (1) - isMqttConnected (0)
14:26:48.870 -> Wifi or MQTT not connected!!!
14:26:52.882 -> MQTT: Connecting to broker @pi4-4zig ... unable to connect, MQTT_CONNECT_FAILED
14:27:02.872 -> MQTT: Connecting to broker @pi4-4zig ... unable to connect, MQTT_CONNECT_FAILED
```

My code is barely more than the sample code :- 

```
/*
  SimpleMQTTClient.ino
  The purpose of this exemple is to illustrate a simple handling of MQTT and Wifi connection.
  Once it connects successfully to a Wifi network and a MQTT broker, it subscribe to a topic and send a message to it.
  It will also send a message delayed 5 seconds later.
*/

#define TIME_TO_SLEEP 10

#include "EspMQTTClient.h"
EspMQTTClient client(
  "SSID",
  "PASSWORD",
  "pi4-4zig",  // MQTT Broker server ip
  "MQTTUsername",   // Can be omitted if not needed
  "MQTTPassword",   // Can be omitted if not needed
  "TestClient",     // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted
);

void setup()
{
  Serial.begin(115200);

  // Optionnal functionnalities of EspMQTTClient : 
  client.enableDebuggingMessages(); // Enable debugging messages sent to serial output
  client.enableHTTPWebUpdater(); // Enable the web updater. User and password default to values of MQTTUsername and MQTTPassword. These can be overrited with enableHTTPWebUpdater("user", "password").
  client.enableLastWillMessage("TestClient/lastwill", "I am going offline");  // You can activate the retain flag by setting the third parameter to true
}

// This function is called once everything is connected (Wifi and MQTT)
// WARNING : YOU MUST IMPLEMENT IT IF YOU USE EspMQTTClient
void onConnectionEstablished()
{
  // Subscribe to "mytopic/test" and display received message to Serial
  client.subscribe("mytopic/test", [](const String & payload) {
    Serial.println(payload);
  });

  // Subscribe to "mytopic/wildcardtest/#" and display received message to Serial
  client.subscribe("mytopic/wildcardtest/#", [](const String & topic, const String & payload) {
    Serial.println(topic + ": " + payload);
  });

  // Publish a message to "mytopic/test"
  client.publish("mytopic/test", "This is a message"); // You can activate the retain flag by setting the third parameter to true

  // Execute delayed instructions
  //client.executeDelayed(5 * 1000, []() {
  //  client.publish("mytopic/test", "This is a message sent 5 seconds later");
  //});
  client.executeDelayed(5 * 1000, my_function);
}

void my_function() {
  Serial.println("Enterered my_function()");
  Serial.print("isConnected (" + String(client.isConnected()) + ")");
  Serial.print(" - isWifiConnected (" + String(client.isWifiConnected()) + ")");
  Serial.println(" - isMqttConnected (" + String(client.isMqttConnected()) + ")");

  // IMPORTANT, protect from spawning multiple delayed instructions when the connection is lost
  if (!client.isConnected()) {
    Serial.println("Wifi or MQTT not connected!!!");
    return;
  }
  client.publish("mytopic/test", "This is a message");
  Serial.println("Going to sleep now");
  
  client.executeDelayed(TIME_TO_SLEEP * 1000, my_function);
}
void loop()
{
  client.loop();
}
```
Any ideas how I can fix this?

Thanks

plapointe6, 2020-02-18:
Seem to be an odd situation. Wich board are you using, an ESP8266 or an ESP32 ? Also, I see that after the MQTT disconnect, you are not able to reconnect to it. If you manually restart your ESP after a connection lost, it will connect to it instantly ?

ajaybnl, 2020-08-25:
Set your MQTT Client Identifier to Random Words (TestClient could be the issue)

plapointe6, 2020-09-03:
I am closing this issue as the code changed since you opened it and no anwser was given after my initial comment.

closed 2020-09-03 by plapointe6

dankamongmen, 2022-12-03:
> Set your MQTT Client Identifier to Random Words (TestClient could be the issue)

nothing to do with this particular code, but i found this, and aye, i had two devices with the same MQTT client name. changed one up, and my disconnects went away.