plapointe6/EspMQTTClient #46: broker returns invalid UTF-8 message if lastwilltopic read from variable

Stefan300381, 2020-02-16:
Hi,

probably more of user error than bug :-) I'd like to set the lastwilltopic but I receive an invalid UTF-8 message from mosquitto:

`1581849946: Malformed UTF-8 in will topic string from ContactSensor01, disconnecting.`

This DOES work;
`client.enableLastWillMessage("test", "false", false);`

but the example below does not. Why?

```
char chArTemp[] = "test";
client.enableLastWillMessage(chArTemp, "false", false);
```

Thanks in advance

Stefan300381, 2020-02-18:
This solves the problem:

```
static char chArTemp[] = "test";
client.enableLastWillMessage(chArTemp, "false", false);
```

plapointe6, 2020-02-18:
Maybe that the will topic and message get deleted between `enableLastWillMessage` and `connect`. This could be avoided by declaring your variable global insted of declaring it in a function. Making your variable static will also works.

closed 2020-02-18 by plapointe6