plapointe6/EspMQTTClient #48: mqtt server disconnected and never reconnected after working for a day

ajaybnl, 2020-03-19:
I have used this wonderful library for my Relay (Mqtt control via Android APP).

I am just working fine, but when i woke up in the morning, the mqtt does not work from my app.

I have checked the app connection with pc mqtt.fx and is running fine.

I have Arduino OTA support added, so i can update the code, and the esp8266 works again. 
But next morning, same problem.

What could be the problem. (There's no problem in the sketch (see below))

`#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include "EspMQTTClient.h"

#define DEBUG false

EspMQTTClient client(
  "MY_SSID",
  "MY_PASSWORD",
  "broker.hivemq.com",  // MQTT Broker server ip
  "",   // Can be omitted if not needed
  "",   // Can be omitted if not needed
  "TestClientxxx",     // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted
);

// Relay On or Off  (1 or 0)
#define TOPIC "/feedsxyz/relayon"
//Status of the Relay
#define TOPIC1 "/feedsxyz/relaystatus"
//Time Relay is on (Secs)
#define TOPIC2 "/feedsxyz/relaytime"
//Time Esp is Running (Mins)
#define TOPIC3 "/feedsxyz/runtime"

long timer1 = 0;
long timer2 = 0;
long timer3 = 0;
long sec1 = 0;
unsigned long runtime = 0;

#define RELAY 13

void setup() {

  pinMode(RELAY, OUTPUT);
  digitalWrite(RELAY, LOW);

  if (DEBUG) Serial.begin(115200);
  delay(10);

  ArduinoOTA.setHostname("Motor");
  ArduinoOTA.begin();

  // Optionnal functionnalities of EspMQTTClient :
  client.enableDebuggingMessages(DEBUG); // Enable debugging messages sent to serial output
  //client.enableHTTPWebUpdater(); // Enable the web updater. User and password default to values of MQTTUsername and MQTTPassword. These can be overrited with enableHTTPWebUpdater("user", "password").
  //client.enableLastWillMessage(TOPIC, "0");  // You can activate the retain flag by setting the third parameter to true
}

void loop() {
  ArduinoOTA.handle();
  client.loop();

// Send ESP Runtime Every Minute
 if (((millis() - timer3  ) / 1000) >= 60) {
timer3 = millis();
   runtime=runtime+1;
   client.publish(TOPIC3, (String(runtime)));
 }

//Send Motor Status Every 5 Secs
  if (((millis() - timer2  ) / 1000) >= 5) {
    timer2 = millis();
   
    client.publish(TOPIC1, (digitalRead(RELAY) == HIGH) ? "1" : "0");
  }

  //Relay Safety (Relay Off After 2 Hours of ON Time Max)
  if (((millis() - timer1 ) / 1000) >= 5) {
    timer1 = millis();

    if (digitalRead(RELAY) == HIGH) {
      sec1 = sec1 + 5;
    client.publish(TOPIC2, String(sec1));
    } else {
      sec1 = 0;
    }
    if (sec1 >= 7200) {
      sec1 = 0;
      digitalWrite(RELAY, LOW);
      delay(100);
      client.publish(TOPIC1, "0");
    }

  }
}

void onConnectionEstablished()
{
  // Subscribe to "mytopic/test" and display received message to Serial
  client.subscribe(TOPIC, [](const String & payload) {

//Convert String to Char*
    char buffer[20];
    payload.toCharArray(buffer, 20);

//Convert Char* to Int
    int n;
    n = atoi(buffer);

//Relay On or Off
    if (n == 1)digitalWrite(RELAY, HIGH); else digitalWrite(RELAY, LOW);

    if (DEBUG) Serial.println(payload);

    if (DEBUG) Serial.println(n);

  // Publish a Relay Status
    client.publish(TOPIC1, (n == 1) ? "1" : "0");

  });

}`

steros76, 2020-03-19:
Surely it is not a solution to your problem, but keep in consideration millis() overflow in your code.

ajaybnl, 2020-03-20:
> 
> 
> Surely it is not a solution to your problem, but keep in consideration millis() overflow in your code.

I have changed long to unsigned long, and  like this :

    if((unsigned long)(millis() - time_now) > 5000)

Lets See.

ajaybnl, 2020-03-24:
> Surely it is not a solution to your problem, but keep in consideration millis() overflow in your code.

The Problem is solved by changing timer to unsigned long, and Slow Enabling Esp8266 when powering up. (adding a 10uf capacitor to ground and EN with 100k resistor to vcc)

The Esp8266 Based Relay is sucessfully running from many days without hanging.

Thanks for your reply.

closed 2020-03-24 by ajaybnl