plapointe6/EspMQTTClient #50: get matched Topic when wildcard used

pkoevesdi, 2020-05-24:
If I subscribe to a topic with a wildcard:
```
client.subscribe("test/#", [] (const String & payload)  {
      Serial.println(payload);
  });
```
How can I get the full topic (without wildcard) of the matched MQTT message?
Is it a missing feature or a missing documentation?

pkoevesdi, 2020-05-25:
Ok, I found it in examples/SimpleMQTTClient/SimpleMQTTClient.ino. It works like this:
```
client.subscribe("mytopic/wildcardtest/#", [](const String & topic, const String & payload) {
  Serial.println(topic + ": " + payload);
});
```
Still, Maybe it should go into the README.md. I can't seem to find any specification for the messageReceivedCallback fucntion. Are there other interesting parameters? I'd be interested in getting the device that sent the message.

plapointe6, 2020-09-03:
Thanks for pointing this, I updated the readme

closed 2020-09-03 by plapointe6