plapointe6/EspMQTTClient #51: WLAN connection bug on some Fritz!Box routers with ESP32 hardware

Mr-Bubbles, 2020-06-15:
Hello,
there is a rather nasty bug with the WLAN connection when trying to connect to a Fritz!Box router with an ESP32: the connection always fails on the first connection attempt an only works after trying a second time. 
This is not a bug in your code in itself, but one rather deep inside the ESP32 librarys. However, EspMQTTClient is also suffering from it. At the moment there is only a workaround that you could maybe implement in EspMQTTClient?
I tried with a local copy of your library an it works for me at least:
```c++
// ================== Private functions ====================-

void EspMQTTClient::connectToWifi()
{
  WiFi.mode(WIFI_STA);
  #ifdef ESP32
    WiFi.setHostname(mMqttClientName);
  #else
    WiFi.hostname(mMqttClientName);
  #endif
  WiFi.begin(mWifiSsid, mWifiPassword);

  if (mEnableSerialLogs)
    Serial.printf("\nWiFi: Connecting to %s ", mWifiSsid);

  // Workaround to a rather nasty BUG with Fritz!Box routers from AVM, where only every other WLAN connection attempt is working.
  // So we will try up to 3 times.
  int tryCount = 0;
  while (WiFi.status() != WL_CONNECTED && tryCount <= 12)
  {
    Serial.print(".");
    tryCount++;
    delay(500);
    if (WiFi.status() == WL_CONNECT_FAILED && (tryCount % 4 == 0))
    {
      WiFi.begin(mWifiSsid, mWifiPassword);
    }
  }
  Serial.println();

  mLastWifiConnectionAttemptMillis = millis();
}
```
see also here: [ESP32 WiFi.begin works only every second time - workaround](https://github.com/espressif/arduino-esp32/issues/2501)

fah, 2020-07-11:
See #54 for a possible solution.

plapointe6, 2020-09-03:
I am closing this issue as the code has been improved (will be part of v1.9.0 soon). However, @fah pointed interesting changes in #54 that I will look further.

closed 2020-09-03 by plapointe6