plapointe6/EspMQTTClient #53: client cant get payload

jackerfonter, 2020-06-25:
when i publish payload with 

`'{"type":"rawir","rawLr":"9020,4360,738,476,716,1566,744,450,718,1564,718,1564,718,476,718,1564,718,476,716,476,718,1564,718,1564,718,1564,718,476,718,474,718,478,716,478,718,478,716,478,718,476,718,476,718,476,718,1564,692,502,718,476,718,476,718,476,718,456,738,476,716,1566,718,476,718,1564,744,450,718,476,744,1538"}'

the nodemcu cant get the payload  .

but when i  publish with payload
``'{"type":"rawir","rawLr":""}'  then its okay.

whats the problem?
`
`

githubbuh, 2020-06-25:
Did you increase the size of the message buffer?  The example you provided is 320 characters.

plapointe6, 2020-09-03:
Totally true, the default maximum message size is 128.

With version 1.9 that will be releashed soon, you will be able to call `client.setMaxPacketSize(512)` and your exemple should work.

closed 2020-09-03 by plapointe6