plapointe6/EspMQTTClient #55: how do i pass a function to subscribe?

nohkumado, 2020-07-16:
Hello
i tryed the example, and it works so far, even if i don't understand whats done at the subscribe stage....
look like some stuff is created on the fly, but i want to regroup the handling of incoming messages in a function, but i don't see how to pass that function to the subscribe....

i looked at the source, and there a MessageReceivedCallbackWithTopic was used, but i didn't found where that thing is declared nor how to use it?

thanks for your help!

githubbuh, 2020-07-19:
nohkumado,

Hope this helps.

i call my payload processor topicPayloadProcess()

`// onConnectionEstablished() is called just once after both 
// the WiFi client and the MQTT are successfully connected
//
// it is in this function that the subscriber is initialized
//
// when the subscriber is initialized, it is possible to 
// assign a function to process the topic payload
//
// NOTE : If this function is not implemented, then no subscriber
//        will be initialized, which means that this sketch will NOT
//        process incoming MQTT messages
void onConnectionEstablished()
{   
    
  // Subscribe to topic and provide the function that will
  // process the payload
  mqttClient.subscribe(TOPIC_EVENT, [](const String & payload) 
  {
      topicPayloadProcess(payload);
  });
}

// ------------------------------------------------------
// function to process mqtt topic
//
// note that this function is connected to the subscriber
// in onConnectionEstablished
// ------------------------------------------------------
void topicPayloadProcess(String payload)
{
   //do your topic processing here.
}

`

plapointe6, 2020-09-03:
Hello, please see the new updated documentation from the readme. I added things that will anwser your questions.

closed 2020-09-03 by plapointe6