plapointe6/EspMQTTClient #56: Connection to WiFI Access Point (AP) happen once of two in a network with two AP with the same SSID

pberna67, 2020-07-30:
Dear all

Using the SimpleMQTTClient example provided of your project, in some home network can happen that two or more AP with the same SSID are in place. It is important that the library selects the best AP (the one with the strongest signal). I've noticed, however, that using the https://github.com/espressif/arduino-esp32/archive/master.zip ES32 library and your additional library, it doesn't happe,  for example when there are two WiFi Access Point with the **same** SSID.  After the first hw reset of the NINA-W102 SDK the board connects to the AP, after the second no, after the third reset, yes and so on. So your example works once out of two, connecting the board  to the WiFi Network

Any idea why this happen and do you suggest any workaround ?

Thanks

plapointe6, 2020-09-03:
Hello, I am sorry but unfortunately it seems to be an issue with the ESP32 framework more than this particular library. Please refer to the according place to raise the issue. Regards.

closed 2020-09-03 by plapointe6

framps, 2020-09-05:
I noticed the same behavior with version 1.8 (See #60). But it works fine with version 1.9 :wink: