plapointe6/EspMQTTClient #57: MQTT PING

pberna67, 2020-08-06:
I'm using the library with a charging MQTT broker, that charges the MQTT_PINGREQ message too. I recommend to make possible, at the application layer, to configure the rate of the MQTT_PINGREQ .
Anyone know where to configure it in the present library version ?

plapointe6, 2020-09-03:
Hello @pberna67,

Is your needs are linked whit the `setKeepAlive()` function of Pubsubclient ? 
https://github.com/knolleary/pubsubclient

`The keepalive interval is set to 15 seconds by default. This is configurable via MQTT_KEEPALIVE in PubSubClient.h or can be changed by calling PubSubClient::setKeepAlive(keepAlive).`

If yes, I will provide a function from `EspMQTTClient` to set this.

Regards

plapointe6, 2020-09-03:
I added the setKeepAlive method.

closed 2020-09-03 by plapointe6