plapointe6/EspMQTTClient #59: (ESP01 and ESP12E) Disconnects Permanently from Broker after some days (And FIX)

ajaybnl, 2020-08-25:
I have a Water Pump, Street Light working with esp8266 (This Library).

It runs and blinks led light every 2 seconds (for activity tracking).

They run 24x7 in my house without any issue. But after a day or two , it does not listens my mqtt commands (But esp is running, check using activity Led)

I added a fix (Reset Mqtt Connection Every Hour).

It works very good , now it don't quit mqtt broker permanently (for weeks).

Here's the fix , if you run your board for 24x7, then you should add it to your sketch.

Fix:

**Open espmqttclient.h and move these two functions from Private: to Public:**

void connectToWifi();
void connectToMqttBroker();

**Define timer4 variable as unsigned long before setup
then, Add these lines to your sketch:**

 if ((((unsigned long)(millis() - timer4)) ) >= (3600*1000)) {
      timer4 = millis();
    client.connectToWifi();
    client.connectToMqttBroker();
    }

**Problem Fixed.**

githubbuh, 2020-08-25:
Ajay,

I used to have the same problem. Connection dropped but heartbeat LED would still blink. I found that restarting the board nightly solves the problem.

ajaybnl, 2020-09-03:
I have found that resetting my wifi router forces the library to re init mqtt broker

ajaybnl, 2020-09-03:
Wifi modem reset also resets the wifi and mqtt. It works after that.

plapointe6, 2020-09-03:
Great suggestion. how do you reset your board/wifi modem ? Are you doing it programatically ? I yes maybe that I could be an improvment to the lib to allow automatic restart each X hours.

ajaybnl, 2020-09-04:
I have added a two hour reset routine, just resets ESP.reset();
Like this

Now it is working ok.

plapointe6, 2020-09-07:
I will keep this in mind. For now, I wont add any ESP.reset() in the lib as the version 1.9.0 should fix most of the connection issues with ESP32. Regards

ajaybnl, 2020-09-08:
On Esp8266, i have added a 2 minute isConnected check and reset function, and now it immidiately recovers from lost mqtt states. Now you can safely implement on .cpp.

Check every 30 seconds:
If(!client.isConnected) count++; else count=0;
If(count>4)ESP.reset();

plapointe6, 2020-09-08:
I am not doing that in my sketches and all my ESP8266 recover automatically when the connection is lost from WIFI or MQTT. If you needs to do this there is an issue with your use case, either in your sketch or in the lib.

ajaybnl, 2020-09-08:
> I am not doing that in my sketches and all my ESP8266 recover automatically when the connection is lost from WIFI or MQTT. If you needs to do this there is an issue with your use case, either in your sketch or in the lib. Further investigation must be done, please try without doing that with the latest version and open a new issue with details if you stiil have issues with connection stability.

I have been running your basic sketch with a serial logger all day on esp8266 today (after updating ur library and pubsubclient )
And found the SAME old Always Disconnected (But Trying) Mqtt Connection after long 6.8 hours of successful running and reconnecting approach.

Means it still will stop your gadget not to listen mqtt commands. The final solution would be the already proposed one. ( I will send u serial log with periodic millis showing)

plapointe6, 2020-09-08:
Ok thanks, serail logs would helps as well as your board model. Regards.

ajaybnl, 2020-09-08:
Dear, i'm only ensuring intterupt free esp operation.
Why dont you add my reset code in a example?
You can even solve the mqtt freeze problem by disconnecting and reconnecting wifi ( Tested )

plapointe6, 2020-09-08:
Because I try to understand why this is happening in your side before doing so.

Sometimes, resetting the ESP could cause issue in specific uses cases, if I am able to fix the disconnexion you are experiancing without doing an ESP.reset() it would be great. Else I will put a note in the readme to add your fix.

ajaybnl, 2020-09-08:
Logs: https://pastebin.com/WsiYRbAS

ajaybnl, 2020-09-08:
I am using this test on D1 mini.

plapointe6, 2020-09-08:
Your logs are really useful, thanks !
They gave me some important clues to investigate this issue.
I need to do some debugging, will keep you in track.

ajaybnl, 2020-09-08:
The Test Runs for 9:30 Hours (Check Millis/1000 in Serial Output)
The Board Disconnected and Reconnected Mqtt For About 7 Hours and then it stopped for 43 minutes, until i reset the wifi modem.

ajaybnl, 2020-09-08:
Thanks, I will do a similer test tomorrow also. (But with (! isConnected) > Wifi.Disconnect Fix)

plapointe6, 2020-09-08:
I done some modifications : 
1. Force the MQTT broker disconnection when it fails connecting.
2. Don't allow publishing, subscribing or unsubscibing if the MQTT broker is not connected.
3. Apply your ESP.reset() fix if the MQTT reconnection mechansim fail after 8 attempts (2 minutes) 

These modifications are untested, you can find the changed code here : https://github.com/plapointe6/EspMQTTClient/tree/ajaybnl-fix

If you redo some tests tomorrow, it would be great if you test the updated code from the ajaybnl-fix branch. I want to know if the fixes 1 and 2 will make the MQTT reconnection work. I am curious to see if ESP.reset() will be called even if I applyed the fixes 1 and 2.

ajaybnl, 2020-09-08:
Ok i will do it from ajaybnl-fix, but add a wifi disconnect also to the try's.

plapointe6, 2020-09-08:
Good idea, I added that

ajaybnl, 2020-09-10:
Last day i ran 2 test devices. One with broker.Hivemq.com , Another with my local raspberry pi 4 mqtt broker, 
Hivemq: https://pastebin.com/2iqndVBB
Local: https://pastebin.com/wbW7h5kx

plapointe6, 2020-09-10:
Thanks for doing these tests.

With broker.Hivemq.com, there was many connection timout errors in the night despites the restart of the board/wifi (maybe there was an issue with broker.Hivemq.com at this time, maybe not ?). However it managed to connect after some time. I think that it could't hurt to implement such failure protection (I mean wifi/board restart) in such cases.

ajaybnl, 2020-09-10:
The ajaybnl-fix worked (Infact on local mqtt server)
I think you should add the wifi reconnect hack for your master repo and just add a optional client.resetOnLongDisconnectedStates(True) Function for people like me.

ajaybnl, 2020-09-10:
Btw you did the ajaybnl-fix just flawlessly. You are awesome coder.

plapointe6, 2020-09-10:
> I think you should add the wifi reconnect hack for your master repo and just add a optional client.resetOnLongDisconnectedStates(True) Function for people like me.

It's exactly what I had in mind 👍 
I will do so soon and merge with master.

> Btw you did the ajaybnl-fix just flawlessly. You are awesome coder.

Thanks !

ajaybnl, 2020-09-12:
Then please do it, and close this issue.

ajaybnl, 2020-09-14:
The ajaybnl-fix failed today for my 1 board : It does not subscribe to any topic. My board was flashed many times with ajaybnl-fix, but never responded, but sending mqtt messages.
Until i revert the library back to master. everything worked back to normal. 

Please fix your ajaybnl-fix script before updating to your master.

plapointe6, 2020-09-14:
It's why I haven't merged yet. Your logs showed many"MQTT! Trying to subscribe when disconnected, skipping." and I think that there is a timing issue between callbacks. I will address that before merging to master.

plapointe6, 2020-09-16:
Merged to master. 
I will do some testing an I will release that with 1.10.0 soon.

plapointe6, 2020-09-16:
Releashed in 1.10.0

closed 2020-09-16 by plapointe6