plapointe6/EspMQTTClient #60: Client fails to connect to broker if WLAN connect fails in first client.loop() call

framps, 2020-09-04:
It took me some time to isolate the root cause the last days. For unknown reasons I was only able to connect to the broker in every second boot of my ESP32. I instrumented the client with some debug statements and finally found out [this test](https://github.com/plapointe6/EspMQTTClient/blob/1faf534367f27e37c26faec8cdde1d0d192dad0c/src/EspMQTTClient.cpp#L194) does not handle a WLAN setup and no retry is executed.

I commented out the test such that a connect was executed all the time when there is no connection and then my ESP was able to connect to the broker all the time.

I'm sure the if statement handles some retry and cannot be commented out. 

See following debug log when 
```
19:19:03.476 -> WiFi: NOT CONNECTED
19:19:03.476 -> WiFi: RETRY
19:19:03.508 -> WiFi: MyWLAN 910 0 - 
19:19:04.174 -> [ 3354 ms ] IR: 65535  Full: 65535  Visible: 0  Lux: -1.000000
19:19:04.240 -> MQTT! publish failed, is the message too long ?
19:19:05.169 -> WiFi: LOOP
19:19:05.169 -> WiFi: NOT CONNECTED
19:19:05.203 -> WiFi: RETRY
19:19:05.203 -> WiFi: MyWLAN 910 0 - 
19:19:05.900 -> [ 5076 ms ] IR: 65535  Full: 65535  Visible: 0  Lux: -1.000000
19:19:05.966 -> MQTT! publish failed, is the message too long ?
19:19:06.897 -> WiFi: LOOP
19:19:06.897 -> WiFi: NOT CONNECTED
19:19:06.930 -> WiFi: RETRY
19:19:06.930 -> WiFi: MyWLAN 910 0 - 
....
```

See following log when there is no issue:
```
19:21:32.999 -> WiFi: LOOP
19:21:32.999 -> WiFi: CONNECTED
19:21:32.999 -> WiFi: Connected, ip : 192.168.0.116
19:21:33.065 -> MQTT: Connecting to broker @192.168.0.12 ... ok.
19:21:33.098 -> MQTT << [mytopic/test] This is a message
19:21:33.729 -> [ 3381 ms ] IR: 65535  Full: 65535  Visible: 0  Lux: -1.000000
19:21:33.795 -> MQTT << [/dawn/sensor/2] 65535;65535;0;-1.000000
19:21:34.725 -> WiFi: LOOP
19:21:34.758 -> WiFi: CONNECTED
19:21:35.455 -> [ 5105 ms ] IR: 65535  Full: 65535  Visible: 0  Lux: -1.000000
...
```

framps, 2020-09-05:
Just noticed the code I have in my Arduino is 1.8 whereas the latest code now is 1.9. I updated the code in my local Arduino and the issue is gone now :+1: 

I have no idea how the code is rolled in into Arduino. This would have saved me a lot of debugging time :wink: .

plapointe6, 2020-09-07:
Glad to hear that the recent 1.9 update fixed your issue 👍

closed 2020-09-07 by plapointe6