plapointe6/EspMQTTClient #64: Version 1.10 in Arduino Library Fails

dansteingart, 2020-10-16:
Versions 1.08 and 1.09 compile and work well (thanks!), but 1.10 fails with:

```
/Users/foo/Documents/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp: In member function 'void EspMQTTClient::loop()':
/Users/foo/Documents/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp:272:13: error: 'class EspClass' has no member named 'reset'
         ESP.reset();
```

plapointe6, 2020-10-16:
What is your board ?
I released a fix in 1.10.1 that was supposed to fix this issue, can you tell me if you are using version 1.10.1 ?
Regards

dansteingart, 2020-10-20:
The Heltec WiFiKit 32. The 1.10.1 release wasn’t showing up in the 
Arduino library manager, just 1.10.0  thx.

![](cid:FB56DE13-E4E5-47CC-91D8-8614023E96F8@gmail.com 
"PastedImage.png")

Shtaiga, 2020-10-20:
Have problem with 1.10.1

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'bool EspMQTTClient::setMaxPacketSize(uint16_t)':

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:342:30: error: 'class PubSubClient' has no member named 'setBufferSize'

   bool success = _mqttClient.setBufferSize(size);

                              ^

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::setKeepAlive(uint16_t)':

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:457:15: error: 'class PubSubClient' has no member named 'setKeepAlive'

   _mqttClient.setKeepAlive(keepAliveSeconds);

plapointe6, 2020-10-21:
@Shtaiga, you need to update the PubSubClient library to his latest version to make it work ;)

@dansteingart, it's a bit odd that version 1.10.1 does not show up in the library manager, I will need to investigate that further.

Appendit, 2020-10-21:
@plapointe6 the library.json has "version": "1.10.0" in it. thats probably why

haruny, 2020-11-08:
I'm having the same issue on 1.10.2. I had to downgrade all the way to 1.8.0.
I'm using a lolin d32.

plapointe6, 2020-11-08:
> I'm having the same issue on 1.10.2. I had to downgrade all the way to 1.8.0.
> I'm using a lolin d32.

Hello,
You had the same problem as "Shtaiga" with 1.10.2 ?
If so, it's because PubSubClient library must be upgraded to his latest version.

haruny, 2020-11-08:
@plapointe6 I had a fresh install of the latest arduino and received the same error message as above.

plapointe6, 2020-11-08:
@haruny 

Do you have this error : 

```
/Users/foo/Documents/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp: In member function 'void EspMQTTClient::loop()':
/Users/foo/Documents/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp:272:13: error: 'class EspClass' has no member named 'reset'
         ESP.reset();
```

Or this error : 

```
E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'bool EspMQTTClient::setMaxPacketSize(uint16_t)':

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:342:30: error: 'class PubSubClient' has no member named 'setBufferSize'

bool success = _mqttClient.setBufferSize(size);

                          ^
E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::setKeepAlive(uint16_t)':

E:\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:457:15: error: 'class PubSubClient' has no member named 'setKeepAlive'

_mqttClient.setKeepAlive(keepAliveSeconds);
```

I need to know exactly what is the error you encounter.

thanks

haruny, 2020-11-09:
Sorry for not being precise.
here's the exact text I'm getting.

```
Arduino: 1.8.13 (Windows 10), Board: "LOLIN D32, Default, 80MHz, 921600, Verbose"

C:\Users\harun\OneDrive\Documents\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'bool EspMQTTClient::setMaxPacketSize(uint16_t)':

C:\Users\harun\OneDrive\Documents\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:342:30: error: 'class PubSubClient' has no member named 'setBufferSize'

   bool success = _mqttClient.setBufferSize(size);
                              ^
C:\Users\harun\OneDrive\Documents\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::setKeepAlive(uint16_t)':
C:\Users\harun\OneDrive\Documents\Arduino\libraries\EspMQTTClient\src\EspMQTTClient.cpp:457:15: error: 'class PubSubClient' has no member named 'setKeepAlive'
   _mqttClient.setKeepAlive(keepAliveSeconds);
               ^
Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\harun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: D:\Prg\Arduino\libraries\WiFi
exit status 1

Error compiling for board LOLIN D32.
```

haruny, 2020-11-09:
I confirm when I upgraded the Pubsub client, the problem went away. sorry for false alarm.

plapointe6, 2020-11-09:
Good to hear

closed 2020-11-09 by plapointe6