plapointe6/EspMQTTClient #65: MQTT connection handling only doesn't work

G47S53, 2020-10-22:
Hello,

I'm trying to use your library without the wifi connection because I want to use the OTA and a fixed ip for my project.

But the connection with mqtt server doesn't work with:
EspMQTTClient ClientMqtt (mqttServerIp, mqttServerPort, mqttUsername, mqttPassword, mqttClientName);

The Wifi connection is ok, because I have access to the NTP server for my project.

If i use the line:
EspMQTTClient ClientMqtt(ssid, password, mqttServerIp, mqttUsername, mqttPassword, mqttClientName, mqttServerPort);
It's work fine, but i have a dynamic IP for my esp8266 and i want a static ip...

Do you have an idea

Best regards

zzpaf, 2020-10-28:
Same issue here, as a workaround, you can go down to 1.8.0 version, functionality seems broken since 1.9 onwards

plapointe6, 2020-10-28:
I will do some tests and correct this functionnality. I will let you know when it will be solved. Thanks for reporting.

G47S53, 2020-10-28:
For working, I used the trick from this post: https://github.com/plapointe6/EspMQTTClient/issues/59

I move the functions connectToWifi() and connectToMqttBroker() from Private to Public in espmqttclient.h. Access to these functions is very practical in the sketch. Maybe a modification to be made in your next version?

plapointe6, 2020-11-09:
Should be resolved with the fix I pushed yesterday. This fix has not been releashed yet because it is in testing phase. If someone want to test the new code, please update the library manually with the latest master branch code.

plapointe6, 2020-11-14:
Fix published with release 1.11.0

closed 2020-11-14 by plapointe6

zzpaf, 2020-12-15:
I believe a new issue has arisen from this issue. On version 1.11.1 if use connection to mqtt without management of wifi, if mqtt can't connect after a while it also disconnects the wifi despite it is not supposed to be managing the wifi.

If we go down to 1.8.0 this does not happen, mqtt just keeps trying to connect. Could this be related?

plapointe6, 2020-12-15:
Thanks for reporting, can you please open a new issue ? It will be simpler for me to track the issue you just reported if you do so.