plapointe6/EspMQTTClient #66: Connect to MQTT broker with hostname

fredericabiven, 2020-11-02:
Hello,

I am trying to connect to my MQTT server hosted on my Raspberry Pi. Since it is hosted on a private network, I would like to connect to the MQTT server with the RPI hostname (raspberrypi.local). I know I can manually write its IP address but it is not very convenient in my case (if the IP address of the RPI changes).

I was able to ping the "raspberrypi.local" hostname from a MacBook, so I know it's available on my private network.

Is there any way I can connect with the hostname with this library?

The code for creating the client :
`client = new EspMQTTClient(
    "ssid",
    "password",
    "raspberrypi.local",
    "rpi", 
    1883
  );`

And the Serial Monitor :
`MQTT: Connecting to broker @raspberrypi.local ... unable to connect, MQTT_CONNECT_FAILED
`

Thanks!

plapointe6, 2020-11-02:
Hello,

Connecting to a broker with the hostname should already work, I have seen some doing it.

- Be sure to supply an unique client name when connecting (in your exemple, the client name is "rpi"). This uniquly identify your MQTT client.

- You could try with the IP if it work.

- If you have an ESP32, you could try to ping raspberrypi.local with the help of this : https://techtutorialsx.com/2019/10/05/esp32-arduino-pinging-a-remote-host/
- Here is the link for an ESP8266 : https://github.com/dancol90/ESP8266Ping

If you are able to connect to your broker with the ip and you are also able to ping raspberrypi.local with your board, I will do more testing in my side.

fredericabiven, 2020-11-02:
Thank you for the quick response! 

Interesting enough, by doing quick tests with that Ping library you sent me, I was able to find out that ESP32 doesn't like the ".local" part. I was able to Ping with the hostname "raspberrypi" instead of "raspberrypi.local".

It does also work with your library. Great job on your work, it is very useful for my personal project.

Many thanks!

plapointe6, 2020-11-02:
Glad to hear it :)

closed 2020-11-02 by plapointe6