plapointe6/EspMQTTClient #67: Connection to WiFI Access Point happen once of two in a network with two AP

pberna67, 2020-11-05:
Dear Developer,

I've encountered Wifi connection problems using the library network with a specific SSID. In some home network can happen that two or more AP with the same SSID are in place. There are different scenarios: the two access points operate at the same frequency band and are connected together with an ethernet cable (Wifi range extension), but can happen, more frequently, that they operate in the same frequency band 

It is important that the library selects the best AP (the one with the strongest signal). I've noticed, however that using the https://github.com/espressif/arduino-esp32/archive/master.zip ES32 library and  EspMQTTClient lib, it doesn't happen,  and to connect to two Access point with the same SSID, the below example works once out of two (if the Access point operate in two different frequency band (100% reproducibility), or it can, sometime not be able to connect to the Wifi network (if the two access points operate in the same frequency band). After the first hw reset of the NINA-W102 EVK (ESP 32) the board connects the AP, after the second no, after the third reset, yes and so on. 
I think the problem is in the native Wifi.begin method, but I'm not sure.

If there is only one access point in the network no connectivity problem is detected.

**Setup** 
https://github.com/espressif/arduino-esp32/archive/master.zip ES32   HAL
EspMQTTClient lib 1.10.1 version
SimpleMQTTClient   example related to the EspMQTTClient 

**Hw**
NINA-W101 EVK, but also ESP6832 board with related HAL

Test report with two access points which operate in the same frequency band with the same SSID. When the WiFi module disconnect it is not able to recover the connection.

WEB: Updater ready, open http://device:7c11f525-c2c8-4f04-b09e-68c5c3c4e256.local in your browser and login with username '2UGGED3TVEPBIYJUGY84' and password 'mnuJ7c8myednY27HHzlk3TJOzyXRPXE43Z76UDRC'.
20:34:02.164 -> MQTT: Connecting to broker "mqtt.broker" with client name "xxxxxxx )unable to connect (16.494000s), reason: MQTT_CONNECTION_TIMEOUT
20:34:17.284 -> MQTT: Retrying to connect in 15 seconds.
20:34:17.324 -> MQTT!: Failed MQTT connection count: 1 
20:34:32.304 -> MQTT: Connecting to broker "mqtt.thingstream.io" with client name "devicexxxx" ... (31.495000s) - ok. (31.834000s) 
20:34:32.624 -> MQTT: Subscribed to [mytopic/test]
20:34:32.624 -> MQTT: Subscribed to [mytopic/wildcardtest/#]
20:34:32.664 -> MQTT << [mytopic/test] This is a message
20:34:33.024 -> MQTT >> [mytopic/test] This is a message
20:34:33.024 -> This is a message
20:34:37.629 -> MQTT << [mytopic/test] This is a message sent 5 seconds later
20:34:37.834 -> MQTT >> [mytopic/test] This is a message sent 5 seconds later
20:34:37.834 -> This is a message sent 5 seconds later
20:34:42.264 -> ets Jun  8 2016 00:22:57
20:34:42.264 -> 
20:34:42.264 -> rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
20:34:42.264 -> configsip: 0, SPIWP:0xee
20:34:42.264 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:34:42.264 -> mode:DIO, clock div:2
20:34:42.264 -> load:0x3fff0018,len:4
20:34:42.264 -> load:0x3fff001c,len:1044
20:34:42.264 -> load:0x40078000,len:10044
20:34:42.264 -> load:0x40080400,len:5872
20:34:42.264 -> entry 0x400806ac
20:34:42.624 -> ⸮
20:34:43.264 -> WiFi: Connecting to Vodafone-27586605 ... (0.651000s) 
20:34:43.384 -> WiFi! Connection attempt failed, delay expired. (0.781000s). 
20:34:43.864 -> E (1961) wifi: esp_wifi_disconnect 1153 wifi not start
20:34:43.904 -> 
20:34:43.904 -> WiFi: Connecting to Vodafone-27586605 ... (1.284000s) 
20:35:43.884 -> WiFi! Connection attempt failed, delay expired. (61.285000s). 
20:35:43.884 -> E (61962) wifi: esp_wifi_disconnect 1153 wifi not start
20:35:44.404 -> E (62462) wifi: esp_wifi_disconnect 1153 wifi not start

plapointe6, 2020-11-08:
Hello,

I see some "E (61962) wifi: esp_wifi_disconnect 1153 wifi not start" errors.
Like discussed in issue #69, can you please redo some testing with the latest ESP32 core (1.0.4).

Thanks !

plapointe6, 2020-11-14:
Hello, I see that updating EP32 core resolved your issue #69, did it resolved this issue too ?

plapointe6, 2020-12-15:
No feedback after a month, it's probably resolved since the update of ESP32 core. Closing

closed 2020-12-15 by plapointe6