plapointe6/EspMQTTClient #69: EspMQTTClient unable to recconnect if the Access Point is switched off that on

pberna67, 2020-11-08:
Using SimpleMQTTClient.ino example, configured to be connected to any MQTT broker (But is not broker dependent)

EspMQTTClient client(
  "MYSSID",           // AP SSID
  "mypasswor",       // AP password
  "137.135.83.217", // MQTT Broker server ip, mqtt.eclipse.org
  "",                         // username: Can be omitted if not needed
  "",                        // password: Can be omitted if not needed 
  "MyClient",         // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted

);  

I tried to verify the error recover capability of the SimpleMQTTClient library. Lunching the example, it connects to the broker, then **I simulate a Wifi failure switching off the Access Point**,  waiting some seconds, then switching on it again. Unfortunately the EspMQTTClient is not able reconnect to the Wifi network.  Only after a Module reset (ESP32 NINA W102) the ESPClinet reconnect

**Why ?**

Report

12:58:31.444 -> ⸮MQTT! Trying to publish when disconnected, skipping.
12:58:32.075 -> 
12:58:32.075 -> WiFi: Connecting to WIND-WiFi-C69F ... (0.651000s) 
12:58:33.312 -> WiFi: Connected (1.904000s), ip : 192.168.10.154 
12:58:33.312 -> WEB: Updater ready, open http://MyClient.local in your browser and login with username '' and password ''.
12:58:33.841 -> MQTT: Connecting to broker "137.135.83.217" with client name "MyClient" ... (2.410000s) - ok. (3.117000s) 
12:58:34.534 -> MQTT: Subscribed to [paolo/door]
12:58:34.534 -> MQTT << [paolo/alarm] Alarm OFF
12:58:35.064 -> MQTT >> [paolo/door] 0
12:58:35.064 -> 0

Nowe the access point is switched off, wait some seconds, than switch on again

12:58:56.033 -> MQTT! Lost connection (24.639000s). 
12:58:56.071 -> MQTT: Retrying to connect in 15 seconds. 
12:58:56.071 -> WiFi! Lost connection (24.640000s). 
12:58:56.071 -> E (25345) wifi: esp_wifi_connect 1145 wifi not start
12:58:56.619 -> E (25878) wifi: esp_wifi_disconnect 1153 wifi not start
12:58:56.619 -> 
12:58:56.619 -> WiFi: Connecting to WIND-WiFi-C69F ... (25.200000s) 
12:59:56.604 -> WiFi! Connection attempt failed, delay expired. (85.200000s). 
12:59:56.604 -> E (85877) wifi: esp_wifi_disconnect 1153 wifi not start
12:59:57.131 -> E (86377) wifi: esp_wifi_disconnect 1153 wifi not start
12:59:57.131 -> 
12:59:57.131 -> WiFi: Connecting to WIND-WiFi-C69F ... (85.700000s) 
13:00:57.112 -> WiFi! Connection attempt failed, delay expired. (145.700000s). 
13:00:57.112 -> E (146378) wifi: esp_wifi_disconnect 1153 wifi not start
13:00:57.614 -> E (146877) wifi: esp_wifi_disconnect 1153 wifi not start
13:00:57.614 -> 
.....
No way to reconnect.

Only after a Module reset (ESP32 NINA W102) the  ESPClinet reconnect

13:08:25.582 -> rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
13:08:25.582 -> configsip: 0, SPIWP:0xee
13:08:25.582 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:08:25.582 -> mode:DIO, clock div:2
13:08:25.582 -> load:0x3fff0018,len:4
13:08:25.582 -> load:0x3fff001c,len:1044
13:08:25.582 -> load:0x40078000,len:10044
13:08:25.582 -> load:0x40080400,len:5872
13:08:25.582 -> entry 0x400806ac
13:08:25.929 -> ⸮MQTT! Trying to publish when disconnected, skipping.
13:08:26.584 -> 
13:08:26.584 -> WiFi: Connecting to WIND-WiFi-C69F ... (0.686000s) 
13:08:28.836 -> WiFi: Connected (2.961000s), ip : 192.168.10.154 
13:08:28.889 -> WEB: Updater ready, open http://MyClient.local in your browser and login with username '' and password ''.
13:08:29.391 -> MQTT: Connecting to broker "137.135.83.217" with client name "MyClient" ... (3.468000s) - ok. (6.945000s) 
13:08:32.848 -> MQTT: Subscribed to [paolo/door]
13:08:32.848 -> MQTT << [paolo/alarm] Alarm OFF
13:08:33.302 -> MQTT >> [paolo/door] 0
13:08:33.302 -> 0

plapointe6, 2020-11-08:
First, I really appreciate your issue description. The more details there is, the easyer it is for me to work on it.

I just made a test with one of my ESP8266 and it is able to reconnect automatically : 

```c++
#include "EspMQTTClient.h"

EspMQTTClient client(
...
);

void setup()
{
  Serial.begin(115200);

  // Optionnal functionnalities of EspMQTTClient : 
  client.enableDebuggingMessages(); // Enable debugging messages sent to serial output
  client.enableHTTPWebUpdater(); // Enable the web updater. User and password default to values of MQTTUsername and MQTTPassword. These can be overrited with enableHTTPWebUpdater("user", "password").
  client.enableLastWillMessage("TestClient/lastwill", "I am going offline");  // You can activate the retain flag by setting the third parameter to true
}

// This function is called once everything is connected (Wifi and MQTT)
// WARNING : YOU MUST IMPLEMENT IT IF YOU USE EspMQTTClient
void onConnectionEstablished()
{
  // Subscribe to "mytopic/test" and display received message to Serial
  client.subscribe("mytopic/test", [](const String & payload) {
    Serial.println(payload);
  });

  // Subscribe to "mytopic/wildcardtest/#" and display received message to Serial
  client.subscribe("mytopic/wildcardtest/#", [](const String & topic, const String & payload) {
    Serial.println(topic + ": " + payload);
  });

  // Publish a message to "mytopic/test"
  client.publish("mytopic/test", "This is a message"); // You can activate the retain flag by setting the third parameter to true

  // Execute delayed instructions
  client.executeDelayed(5 * 1000, []() {
    client.publish("mytopic/test", "This is a message sent 5 seconds later");
  });
}

void loop()
{
  client.loop();
}
```

Logs : 
```
WiFi: Connecting to Nuage ... (0.880000s) 
WiFi: Connected (4.722000s), ip : 192.168.1.112 
WEB: Updater ready, open http://test1.local in your browser and login with username 'pingouin000' and password 'Mqtt8636!'.
MQTT: Connecting to broker "192.168.1.101" with client name "test1" ... (5.227000s) - ok. (5.259000s) 
MQTT: Subscribed to [mytopic/test]
MQTT: Subscribed to [mytopic/wildcardtest/#]
MQTT << [mytopic/test] This is a message
MQTT >> [mytopic/test] This is a message
This is a message
MQTT << [mytopic/test] This is a message sent 5 seconds later
MQTT >> [mytopic/test] This is a message sent 5 seconds later
This is a message sent 5 seconds later

// ----- Here I deactivate my WiFi

WiFi! Lost connection (146.055000s). 
MQTT! Lost connection (146.209000s). 
MQTT: Retrying to connect in 15 seconds. 
WiFi: Connecting to Nuage ... (146.857000s) 

// ----- Here I reconnect my WiFi

WiFi: Connected (182.111000s), ip : 192.168.1.112 
WEB: Updater ready, open http://test1.local in your browser and login with username 'pingouin000' and password 'Mqtt8636!'.
MQTT: Connecting to broker "192.168.1.101" with client name "test1" ... (182.617000s) - ok. (182.645000s) 
MQTT: Subscribed to [mytopic/test]
MQTT: Subscribed to [mytopic/wildcardtest/#]
MQTT << [mytopic/test] This is a message
MQTT >> [mytopic/test] This is a message
This is a message
MQTT << [mytopic/test] This is a message sent 5 seconds later
MQTT >> [mytopic/test] This is a message sent 5 seconds later
This is a message sent 5 seconds later
```

I will try the same thing with an ESP32. I admit that I dont use ESP32 much, so I do less tests with them than with my ESP8266.

I will update in a couple of minutes.

plapointe6, 2020-11-08:
Same thing, it work with my ESP32 (LOLIN D32)

Logs: 
```
WiFi: Connecting to Nuage ... (0.610000s) 
WiFi: Connected (0.768000s), ip : 192.168.1.42 
WEB: Updater ready, open http://test1.local in your browser and login with username '' and password ''.
MQTT: Connecting to broker "192.168.1.101" with client name "test1" ... (1.273000s) - ok. (1.286000s) 
MQTT: Subscribed to [paolo/door]
MQTT << [paolo/alarm] Alarm OFF

// - I disconnect my WiFi

MQTT! Lost connection (24.205000s). 
MQTT: Retrying to connect in 15 seconds. 
WiFi! Lost connection (24.205000s). 

WiFi: Connecting to Nuage ... (24.747000s) 

// - I reconnect my WiFi

WiFi: Connected (44.164000s), ip : 192.168.1.42 
WEB: Updater ready, open http://test1.local in your browser and login with username '' and password ''.
MQTT: Connecting to broker "192.168.1.101" with client name "test1" ... (44.668000s) - ok. (44.690000s) 
MQTT: Subscribed to [paolo/door]
MQTT << [paolo/alarm] Alarm OFF
```

plapointe6, 2020-11-08:
It seem to be related to the ESP32 core and already resolved
See this issue : https://github.com/espressif/arduino-esp32/issues/3522

Can you update the ESP32 core to it latest version and retry please ? I made my test with ESP32 1.0.4

Thanks

pberna67, 2020-11-08:
Thanks for the bast replay! but what to you mean with  "update the ESP32 core to it latest version"
do I need to change the hw platform or what else ? I'm using the NINA W102 ESP32 module by u-blox at the moment but there is no any hw update

pberna67, 2020-11-13:
Yes success, it works with ESP32 1.0.4 !

closed 2020-11-13 by pberna67

pberna67, 2020-11-13:
Success ! It works with  ESP32 1.0.4 library. Thanks !

T94T, 2021-07-21:
Hi,
First of all: Thank you very much for your great library!

We have the exact same problem with the ESP32S2 module.

Using the Arduino esp32 with latest commit https://github.com/espressif/arduino-esp32/commit/e12d8c8ff1a241160fc38d2f805c5ea169be7a5e
```
WiFi: Connected (3.407000s), ip : 192.x.x.x 
MQTT: Connecting to broker "192.x.x.x" with client name "CLIENT" ...
MQTT << [devices/CLIENT/status] connected
```
after losing connection
```
WiFi: Connecting to Wireless ... (306.658000s) 
WiFi! Connection attempt failed, delay expired. (366.658000s). 
WiFi: Connecting to Wireless ... (367.158000s) 
WiFi! Connection attempt failed, delay expired. (427.158000s). 
WiFi: Connecting to Wireless ... (427.658000s) 
```