plapointe6/EspMQTTClient #70: setMaxPacketSize not working

bas-eng, 2020-11-14:
The setMaxPacketSize is not working as expected. You can set this and it will get correctly set inside PubSubClient however on each MQTT meassage the lengt is compared to MQTT_MAX_PACKET_SIZE but shoud be compared to _mqttClient.getBufferSize()

See line 606:
` if (strlen(topic) + length + 9 >= **_mqttClient.getBufferSize()**)
  {
    strTerminationPos = length - 1;

    if (_enableSerialLogs)
      Serial.print("MQTT! Your message may be truncated, please change MQTT_MAX_PACKET_SIZE of PubSubClient.h to a higher value.\n");
  }
  else
    strTerminationPos = length;`

plapointe6, 2020-11-14:
Thanks for reporting, I corrected it and I will release the fixes soon.

plapointe6, 2020-11-14:
Fix published with 1.11.0

closed 2020-11-14 by plapointe6