plapointe6/EspMQTTClient #71: Wildcard in example is not displaying message anymore

deNiklaus, 2020-11-17:
## Description of the problem
Since Version 1.11.0 the wildcard-part of the example from the libary is not displaying messages anymore. 

Topics without wildcard work perfectly.
Subscribing with the wildcard is working (Debugging shows incoming messages).

Problem: The function in the example to print the received topic and the message doesn't do anything.

## Versions

- ESPMQTTClient lib: 1.11.0  &nbsp;&nbsp;&nbsp;   (no problem in 1.10.2)
- PubSubClient lib: 2.8.0
- Arduino ESP32 core: 1.0.4

## Hardware
- Board type : ESP32
- Board model : Dev Module

## C++ code

In the example file "SimpleMQTTClient" the following printing of messages from the wildcard-subscription is not doing a thing:
```c++
  client.subscribe("mytopic/wildcardtest/#", [](const String & topic, const String & payload) {
    Serial.println(topic + ": " + payload);
  });
```

## Logs

```
...
MQTT >> [mytopic/wildcardtest/hello] test-payload-for-problem-topic
MQTT >> [mytopic/test] test-payload-without-wildcard
test-payload-without-wildcard
```
Here you see, both topics were received and are visible with the debugging-option. 
However the wildcard-topic does not trigger its Serial.println().

plapointe6, 2020-11-17:
Thanks for reporting this issue. It seems that the QOS functionnality I added with 1.11.0 broke the wildcard. I will publish a fix soon.

plapointe6, 2020-11-17:
The fix has been published with release 1.11.1. Thanks

closed 2020-11-17 by plapointe6

RobotGrrl, 2020-12-29:
Hi @plapointe6, Is this fix supposed to work by updating the library to v1.11.1 in the Arduino libraries manager? I have just tried it, and still not getting the callback function called on a subscribed topic with a # in it. The messages are received, as visible by the debugging-option, but it never calls the callback function.