plapointe6/EspMQTTClient #72: isMqttConnected returns false always

steinundfloete, 2020-12-02:
## Description of the problem
When asking isMqttConnected in loop(), it returns false always, even after onConnectionEstablished was called.
For testing, I implemented onConnectionLost callback the same way as onConnectionEstablished, this works fine.
Would be a nice idea for the next release :), ... same for WiFi (isWifiConnected() works fine)

## Versions

- ESPMQTTClient lib: 1.11.1
- PubSubClient lib: 2.8.0
- Arduino ESP32 / ESP8622 core: v3.2.3-14-gd3e562907

## Hardware
- Board type : ESP32
- Board model : ESP-WROOM-32

## Logs
## C++ code

steinundfloete, 2020-12-02:
otherwise.... great work, THANKS A LOT  !!!!

plapointe6, 2021-11-30:
May be resolved since the last few releases. You can re-open this issue otherwise.

closed 2021-11-30 by plapointe6