plapointe6/EspMQTTClient #73: Function for changing WiFi credentials

radimsejk, 2020-12-15:
- **Background:**
I am using this library and it is perfect for easy connection to WiFi and MQTT. But my devices are not fixed in one place. Sometimes it is my home network (SSID1 and PASS1) and sometimes it is outside connected to my phone hotspot (SSID2, PASS2). So I have done only declaration of this library `EspMQTTClient* MQTTclient;` without any constructor and in my `void ConnectionCheck(void) ` function which is called once in a second I do `MQTTclient = new EspMQTTClient(SSID1, PASS1......)` and after some time `MQTTclient = new EspMQTTClient(SSID2, PASS2......)` . And this approach works.
- **My problem:**
When it looses connection (maybe I go away from the device with my phone) it crashes. It look like this problem is connected with calling `MQTTclient = new EspMQTTClient(SSID2, PASS2......)` because when I comment this line, it does not crash and reconnects. I think that the `new` somwhere clears some flag or something and after that it the library gets confused.
I am using ESP8266, Arduino 1.8.13, ESP8266 core 2.7.4 and this espMQTTClient library 1.11.1
- **Possible solution:**
It would be great if there would be some function like  **`setWiFiCredentials(SSID, PASS)`**.

plapointe6, 2020-12-15:
You are not the first who request such changes on the lib. I will be out for the next weeks but I will take a look when I will come back. Thanks.

ravng, 2021-01-06:
Would love to see this myself. Thanks for a great lib.

tomskopek, 2021-02-03:
I just opened a PR that should solve this https://github.com/plapointe6/EspMQTTClient/pull/78

Thanks for the library Patrick! By the way, I was happy to see on your github bio that you are in Chicoutimi - I do a yearly trip to Lac St Jean to visit my family. Small world :)

closed 2021-02-08 by plapointe6