plapointe6/EspMQTTClient #76: Please add a "onWiFiConnectionEstablished" callback

ikarisan, 2020-12-23:
I am trying to use you MQTT library together with NTPClient libraries. These libraries require at least to wait for an established WiFi connection, before asking a NTP server. Would it be possible to include a onWiFiConnectionEstablished() method as well? Maybe the MQTT broker is not reachable but the WiFi is already usable. 

A client.getWiFiConnection() would also be nice, because some libraries require to pass a reference to a WiFi or WiFiUdp object.

kintel, 2022-02-04:
I was just looking into this. This should be possible to do yourself using WiFi events, smth. like this for esp32:
```
...
WiFi.onEvent(onWiFiConnectionEstablished, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_GOT_IP)
...

void onWiFiConnectionEstablished(WiFiEvent_t event, WiFiEventInfo_t info) {
}
```

There should be opportunities, in the `dev` version of this library, to add such a convenience function though.

kintel, 2022-02-09:
@plapointe6 what's your opinion on using magic functions like `onConnectionEstablished`, which the user _has_ to implement? If it's just a single function, it's quite user friendly, but it can become a bit messy if there are more signals we'd like to expose. Like WiFi connection status.

Allowing users to register callbacks is more classic, but that could introduce an incompatibility with the current version of the library. Not sure if that's a big issue..