plapointe6/EspMQTTClient #77: MQTT Client (esp32) has exceeded timeout

VK6NX, 2020-12-30:
## Description of the problem
Setup: 
MQTT is running on macOS 10.15.7 (no issues whatsoever with any other clients)
EspMQTTClient is running on ESP32 (different dev board platforms were tested, returning same result).

Symptoms: After connecting to MQTT server, client may go through one or two timeouts/reconnects within first 2-3 minutes from initial connection. Practically client is reconnecting almost immediately, however few  MQTT messages can be lost 

## Versions

- ESPMQTTClient lib: 1.11.1
- PubSubClient lib: 2.8
- Arduino ESP32 / ESP8622 core: -put version number here- (To update it go to the arduino board manager and update ESP32 / ESP8266)

## Hardware
- Board type : ESP32
- Board model : ESP32devkitc_v4 ESP32-WROOM32U / ESP32devkitc_v1 ESP32-WROOM-D

## Logs
MQTT log:
1609337737: New client connected from 192.168.1.236 as esp32-ATU-T (p2, c1, k15).
1609337843: New connection from 192.168.1.236 on port 1883.
1609337875: Client esp32-ATU-T has exceeded timeout, disconnecting.
1609337882: New connection from 192.168.1.236 on port 1883.
1609337882: New client connected from 192.168.1.236 as esp32-ATU-T (p2, c1, k15).
1609337987: Client esp32-ATU-T has exceeded timeout, disconnecting.
1609338034: New connection from 192.168.1.236 on port 1883.
1609338034: New client connected from 192.168.1.236 as esp32-ATU-T (p2, c1, k15).
```
Content of the log here
```

## C++ code
common code from example, no modifications
```

plapointe6, 2021-11-30:
These issues are hard to diagnose. Also, connections issues like this may not come from this library. This issue is one year old, I will assume that it is resolved. If it's not the case, you can let me know.

closed 2021-11-30 by plapointe6

VK6NX, 2021-11-30:
Patrick, this is great move - look at the request a year after it lodged and declare it not existed 👍
Actually, yes, I have resolved the issue. Unfortunately, by using alternative MQTT library.
All the best and thank you for committing anyway.

plapointe6, 2021-11-30:
Hello, I am curious, wich MQTT library do you use instead ?