plapointe6/EspMQTTClient #80: Compilation for Wemos D1 fails.

paysan18, 2021-02-23:
I try to use the EspMQTTclient library on a Wemos D1 mini.
Got the message: Error compiling for board LOLIN(WEMOS) D1 & mini.

Can this problem get solved please?

plapointe6, 2021-02-23:
Hello,
The library compile for me with a D1 Mini. Please give me more details about the error message.

paysan18, 2021-02-23:
Hello Plapointe, this is all I got. I’m running the Arduino 1.8.13 on a Debian OS on a Raspberry-PI4.

paysan18, 2021-02-23:
Patrick, after some further testing I found that all of the various ESP boards are giving this error. 
Again, this is on a Debian Buster version running on a PI4.     IDE 1.8.13.

Groeten, Wim

pauled23, 2021-02-24:
I'm also seeing an error when compiling for ESP-01 when compiling SimpleMQTTClient.ino:

```
/home/USERArduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp: In member function 'void EspMQTTClient::onWiFiConnectionLost()':
/home/USER/Arduino/libraries/EspMQTTClient/src/EspMQTTClient.cpp:358:10: error: 'class MDNSResponder' has no member named 'end'
     MDNS.end();
          ^
exit status 1
Error compiling for board Generic ESP8266 Module.
```

paysan18, 2021-02-24:
So I’m glad that I’m not the only one.

Groeten, Wim

plapointe6, 2021-02-24:
It's quite odd, I am not able to reproduce this issue on my side.

Can you reply with the version of the esp8266 core you use ?

A quick workaround for now could be to open `EspMQTTClient.cpp` and comment line no. 358 `MDNS.end();`

paysan18, 2021-02-25:
Patrick, I replaced my  EsppMQTTClient set and the PuSuBClient with a set from a friend of mine. On his system there is no problem.
He is also running the Arduino 1.8.13 version IDE.
We both run it on a Raspberry PI.

Groeten, Wim

paysan18, 2021-02-25:
Problem solved.
Cause of the problem was pubsubclient was not found.
It was located local. After moving it to the Arduino/libraries/ problem was gone.

plapointe6, 2021-02-25:
Great 👍

closed 2021-02-25 by plapointe6