plapointe6/EspMQTTClient #81: hostname resolution

josefgull, 2021-03-12:
i want to connect via the hostname instad of the ip. but it only works withe ip if use the hostname in my broser i can reach the pc:/(/
works
client=new EspMQTTClient (con.WifiName,con.WifiPass,"192.168.0.102", "", "", con.Nodename, 1883 );
works not
client=new EspMQTTClient (con.WifiName,con.WifiPass,"myhostname", "", "", con.Nodename, 1883 );

githubbuh, 2021-12-30:
plapointe6,

I wonder if a better name for this enhancement would be: Connect with hostname OR ip address because I am sure it is not intended that this enhancement would be a breaking change for all who want to connect using an IP address.

Thanks in advance,

githubbuh

EdJoPaTo, 2021-12-30:
Using a hostname works for me. I have never used IP addresses with this library.
My in-home MQTT Broker uses DHCP and has a hostname which I use to connect to it which works fine. Not sure what the problem is or if I am overlooking something?

Rosen01, 2023-01-26:
I confirm that it's working with the hostname.
Looks like there is a DNS problem in your network.

The "myhostname" is not registred on the referenced DNS on the ESP32, perhaps you use different DNS server for the ESP32 and the myhostname server, or the registred computer has another name in the DNS like "myhostname.local",
If you're have referenced an external DNS on every computer on your network you won't be able to register local computer.

Check the result of the command on the myhostname server and on another computer that is on the same network as the ESP32 :
`nslookup myhostname`

If it's correctly registred on the DNS, you will have a result like :

```
Server: <DNS Server name or IP>
Address: <DNS Server IP>

Name: myhostname
Address: <myhostname IP>
```

I think the issue can be closed.