plapointe6/EspMQTTClient #87: Support a list of wifi networks

autoantwort, 2021-08-01:
Instead of providing one ssid password combination, it would be cool if you could provide a list of combinations. For example for my testing network and for the "production" location. Currently you always have to upload a new sketch every time you switch the location. 

But cool project, simplifies my code a lot!

autoantwort, 2021-08-06:
I already started implementing that, will create a PR once I am finished

plapointe6, 2021-11-30:
I may replace the internal WiFI handling completly with the new EspSimpleWifiHandler from https://github.com/plapointe6/EspSimpleWifiHandler

I like the idea of more than one wifi config. I will take a look into this. I think that something already exists to handle this (WifiMulti).

josefgu, 2022-03-03:
i would love to see that feature too.

mojgithub, 2022-07-26:
What is the status of this feature? Is it already implemented or still in development? 

I would also like this feature. I mean i am moving my device between 4 different locations(wifis).

autoantwort, 2022-07-26:
I started the implementation, but now have switched to esphome and have no interest anymore the finish the implementation. esphome can handle multiple wifis.