plapointe6/EspMQTTClient #93: PlatformIO and Arduino doesnt know new versions

EdJoPaTo, 2021-11-20:
## Description of the problem

PlatformIO and Arduino doesnt know about the latest versions.
Both see 1.11.1 as the latest version.
Not sure if updating the `platform.json` and `platform.properties` files is enough?

## Logs
```bash
pio lib show plapointe6/EspMQTTClient
```

```plaintext
plapointe6/EspMQTTClient
========================
A library that provides a wifi and MQTT connection to an ESP8266/ESP32. This library allow to connect and manage the connection to a wifi network and a MQTT broker. Intended to be used with an ESP8266 an ESP32. Dependecy : PubSubClient library

ID: 5956
Version: 1.11.1, released Tue Nov 17 23:14:49 2020
Manifest: https://raw.githubusercontent.com/plapointe6/EspMQTTClient/master/library.properties
Repository: https://github.com/plapointe6/EspMQTTClient.git

…

Examples
--------
https://dl.registry.platformio.org/examples/12/5a/db2109bfededfb7eab8f82d8c00596d7185b481304eee9401ff12de18a49/twoMQTTClientHandling.ino
https://dl.registry.platformio.org/examples/2b/6f/d343539bf302e60d3dedffb799a9d384fae29d4dd503acd858aa281139ed/SimpleMQTTClient.ino

Versions
--------
1.11.1, released Tue Nov 17 23:14:49 2020
1.11.0, released Sat Nov 14 20:02:32 2020
1.10.2, released Wed Oct 21 20:41:42 2020
1.10.0, released Thu Sep 17 12:53:55 2020
1.9.0, released Thu Sep  3 20:13:40 2020
1.8.0, released Thu Jan 16 15:44:56 2020
1.6.0, released Sun Jul 14 19:29:10 2019
```

plapointe6, 2021-11-20:
I've updated library.json and library.properties
I am not sure if I need to do a new release with these files updated.
We will wait a little. If they don't recognize the new version I will publish a new release.

EdJoPaTo, 2021-11-21:
Both Arduino and PlatformIO are still displaying 1.11.1

EdJoPaTo, 2021-11-22:
Both Arduino and PlatformIO have 1.13.1 now. Thanks!

closed 2021-11-22 by EdJoPaTo