plapointe6/EspMQTTClient #94: I need make Download Esp32MQTTClient

thaynapatricio, 2021-11-23:
Hey!

Can you add the **Esp32MQTTClient** library for me to download? I'm not finding her anywhere.

'cause I have this problem...
device:6:29: fatal error: Esp32MQTTClient.h: No such file or directory
[Warning] Failed to generate IntelliSense configuration.
[Error] Uploading sketch 'device.ino': Exit with code=1

And I have just EspMQTTClient

plapointe6, 2021-11-23:
Esp32MQTTClient does not exist. You need to use EspMQTTClient.h wich support both ESP8266 and ESP32.

closed 2021-11-23 by plapointe6

thaynapatricio, 2021-11-23:
There's actually one version of Azure IoT that I didn't find in my directory :)
[AzureIoT.zip](https://github.com/plapointe6/EspMQTTClient/files/7590878/AzureIoT.zip)
.

plapointe6, 2021-11-23:
Hello,

Sorry, I was a bit confused about this. In fact, I think that your issue is not related to EspMQTTClient.

I took a look to the .zip file you provided and I see an Esp32MQTTCLient. However, it has nothing to do with plapointe6/EspMQTTClient.

The library that you are using is this one: https://github.com/VSChina/ESP32_AzureIoT_Arduino
Please refer to this github repo instead.

Thanks