plapointe6/EspMQTTClient #96: EspMQTTClient disconnects my WiFi if fails to connect to broker

zzpaf, 2021-11-24:
Thanks for the great effort on the library!

I am recently finding an issue (Running on ESP32) where MQTT client is disconnecting my WiFi if it fails to connect to broker:
This is quite an unwanted feature as I don't want the client to handle the WiFi connectivity.

I launch the client this way:

```
  client = new EspMQTTClient(
    "mymqttserver.com", 
    1883,
    "MyDeviceName" 
  );
```

Then if for whatever reason there is a connectivity issue to MQTT server:

11:06:04.729 -> MQTT: Connecting to broker "mymqttserver.com" with client name "MyDeviceName" ... (129.665000s)unable to connect (129.681000s), reason: MQTT_CONNECT_FAILED
11:06:04.729 -> MQTT: Retrying to connect in 15 seconds.
11:06:04.729 -> MQTT!: Failed MQTT connection count: 7 
11:06:19.723 -> MQTT: Connecting to broker "mymqttserver.com" with client name "MyDeviceName" ... (144.681000s)unable to connect (144.692000s), reason: MQTT_CONNECT_FAILED
11:06:19.756 -> MQTT: Retrying to connect in 15 seconds.
11:06:19.756 -> MQTT!: Failed MQTT connection count: 8 
11:06:19.756 -> **MQTT!: Can't connect to broker after too many attempt, resetting WiFi ...**
11:06:19.756 -> **WiFi! Lost connection (144.708000s).** 

And this is the end of it...

Why is WiFi being reset if MQTT fails? I have not setup the client to manage WiFi.
I have been using the library for quite some time and I don't think this was happening in the past...

Thanks!

zzpaf, 2021-11-24:
I saw the issue area:

  ```
    if(_failedMQTTConnectionAttemptCount == 8)
      {
        if (_enableSerialLogs)
          Serial.println("MQTT!: Can't connect to broker after too many attempt, resetting WiFi ...");

        WiFi.disconnect(true);
        MDNS.end();
        _nextWifiConnectionAttemptMillis = millis() + 500;

        if(!_drasticResetOnConnectionFailures)
          _failedMQTTConnectionAttemptCount = 0;
      }
```

There are 2 possibilities, one is to be able to change the reconnection attempts (Maybe add a variable and if it is 0 we continue trying) and another one is to be able to have a function:

```
client->resetWifiIfMQTTConnectionFails = true
```

and change the above to:

  ```
    if(_failedMQTTConnectionAttemptCount == 8)
      {
        if (_enableSerialLogs)
          Serial.println("MQTT!: Can't connect to broker after too many attempt, resetting WiFi ...");

        if(resetWifiIfMQTTConnectionFails)
        {
             WiFi.disconnect(true);
        }
        MDNS.end();
        _nextWifiConnectionAttemptMillis = millis() + 500;

        if(!_drasticResetOnConnectionFailures)
          _failedMQTTConnectionAttemptCount = 0;
      }
```

To allow the end user to decide whether MQTT failure will reset WiFi or not.

Hope this helps

plapointe6, 2021-11-24:
I understand you issue.
It's an easy fix. I will fix that soon. 
Thanks

zzpaf, 2021-11-25:
Thank you, happy to test when implemented

plapointe6, 2021-11-25:
The fix is now applied. I haven't tested it though. Please, give it a try if you can and let me know if it work. If so, I will release the fix to be available in Arduino IDE and PlatformIO library manager. 

Regards.

zzpaf, 2021-11-25:
Confirmed that it is working correctly, now it shows that MQTT fails past the 8th attempt and does not manage the WiFi as expected.

Thanks a lot. Awesome quick fixing!

```
23:50:55.836 -> MQTT: Connecting to broker "mqttserver.com" with client name "Client_Name" ... (386.569000s)unable to connect (386.588000s), reason: MQTT_CONNECT_FAILED
23:50:55.836 -> MQTT: Retrying to connect in 30 seconds.
23:50:55.836 -> MQTT!: Failed MQTT connection count: 7 
23:51:25.855 -> MQTT: Connecting to broker "mqttserver.com" with client name "Client_Name" ... (416.588000s)unable to connect (416.605000s), reason: MQTT_CONNECT_FAILED
23:51:25.855 -> MQTT: Retrying to connect in 30 seconds.
23:51:25.855 -> MQTT!: Failed MQTT connection count: 8 
23:51:55.860 -> MQTT: Connecting to broker "mqttserver.com" with client name "Client_Name" ... (446.605000s)unable to connect (446.627000s), reason: MQTT_CONNECT_FAILED
23:51:55.898 -> MQTT: Retrying to connect in 30 seconds.
23:51:55.898 -> MQTT!: Failed MQTT connection count: 9 
```

Thanks,

closed 2021-11-25 by zzpaf