plapointe6/EspMQTTClient #100: Create mqttClientName from MAC-ID

jokubokla, 2022-01-13:
Just an idea: Instead of creating <b>'mqttClientName'</b> values for a bunch of different nodes manually, how about <b>'Node_BEEF1234CAFE'</b> if the MAC-ID is <b>'BE:EF:12:34:CA:FE'</b> as the <i>default value</i> (with the possibility to override of course). Reason: Putting the same code in various different nodes would be much easier if one doesn't need to change the coding for each single node.

Thank's for the project by the way!
Joerg

plapointe6, 2022-01-13:
Good idea. I will take a look at this the next time I work on this. Thanks.

jokubokla, 2022-01-15:
...again just FYI: If you also intend to tweak your sample code:

#define NODENAME "ESP8266Node22"
#define VAR_LASTWILL NODENAME "/lastwill"
#define VAR_CYCLECNT NODENAME "/var/cycle"
(...)
client.enableLastWillMessage(VAR_LASTWILL, "Sorry I am offline");
(...)
client.publish(VAR_CYCLECNT, "{\"CycleCnt\":0}");
(...)

The #define preprocessor concatenates the NODENAME into the MQTT variable name. This makes it easier to seperate MQTT messages in node-red for example.

Thank's again...
Joerg

closed 2022-01-15 by jokubokla