plapointe6/EspMQTTClient #101: Incomplete string when encrypted buffer is received

bangbaew, 2022-01-26:
## Description of the problem
I encrypt a JSON string from node.js using AES-128 ECB mode and send the encrypted buffer to my esp32 via Mqtt, and decrypt on my esp32 using [esp32-Encrypt](https://github.com/josephpal/esp32-Encrypt), however, the encryption process seems to work fine but when receiving the buffer, sometimes it only receives a partial buffer when sending a specific string.

for example: 
this is an example of a complete buffer, and it's always complete if the origin string looks like that
[image]
this is from mqttx which is also subscribed to the same topic
[image]

but when the origin string url is changed from `https://app.binance.com/qr/dplkff1d79e287474989a576f5b33064a7a7` to `https://app.binance.com/qr/dplk3568fe48d0d24fe083c53efdd066c18b` or some other different strings, the received buffer becomes incomplete, it only received a partial of the encrypted buffer, and always reproducible 
[image]

so I checked the mqttx, the encrypted buffer is complete but the highlighted part is only received by the esp32
[image]

## Versions
- ESPMQTTClient lib: 1.13.3
- PubSubClient lib: 1.13.3
- Arduino ESP32 / ESP8622 core: 1.0.6

## Hardware
- Board type : ESP32
- Board model : ESP-WROOM-32 (ESP32 Dev Module)

## C++ code
Mqtt code is the same as in the readme example

bangbaew, 2022-02-01:
I've tried using PubSubClient, it receives the encrypted message correctly
[image]

but the code is different, it gets message length and loops the array buffer until it completes the length
[image]

bangbaew, 2022-02-01:
I've found the solution for ESPEspMQTTClient, looks like it's the problem with `(char*)payload` and `payloadStr.c_str()` which will unexpectedly terminate the encrypted string, so I had to change from `String payloadStr((char*)payload);` to the loop below, and now it works perfectly!
[image]

However, I'm not sure if this can be the permanent solution to this issue, please find a better solution and release the fix in the next updates, thanks.

EdJoPaTo, 2022-02-01:
> please find a better solution

This is an open source project. Others wont just do the work for you. Do tests and suggest one :)

Strings in C/C++ assume the 0 byte as end of string. Does the encrypted data contain the 0 byte? If so, its probably the reason why this happens. Its probably not a good idea to keep the data as data type String then.

bangbaew, 2022-02-01:
> This is an open source project. Others wont just do the work for you. Do tests and suggest one :)
>
> Strings in C/C++ assume the 0 byte as end of string. Does the encrypted data contain the 0 byte? If [...]

Sorry for that, I mean please suggest me if there's a better implementation to the code, because I don't know if using a for-loop like that will affect the efficiency/performance or not, or is there a way to prevent the 0 byte data to terminating the string.
Thank you for providing the reason why the problem happens.